ROBOTIS-GIT/turtlebot3 #520: Measuring the rotation angle and traveled distance

Adel-kn, 2019-11-28:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

Hello everyone: 
I have a program (python) to move the robot rotational or translational and it works. The movement regarding the hardware constraints is not 100% accurate. My question is how can one measure as accurately as possible the rotated angle or traveled distance to find out the error between the desired movement and the actual one? ,
Thanks in advance

ROBOTIS-Will, 2019-12-03:
@Adel-kn 
The actual pose of the robot can be acquired using localization such as [amcl package](http://wiki.ros.org/amcl) and measure the error from the calculated pose with your program. However, it is difficult to get a 100% accurate pose due to many unexpected parameters.

Adel-kn, 2019-12-04:
@ROBOTIS-Will 
Thanks for your help

closed 2019-12-04 by ROBOTIS-Will