ROBOTIS-GIT/turtlebot3 #521: Error When Deleting Turtlebot Model in Gazebo

ShiJbey, 2019-12-02:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x] Simulation

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [x] Simulation

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: melodic-devel
    - Firmware version: N/A

7. Write down the commands you used in order

    - ```roslaunch turtlebot3_gazebo turtlebot3_empty_world.launch```
    - (in second terminal) ```rosservice call gazebo/delete_model '{model_name: turtlebot3_waffle_pi}'```

8. Copy and Paste your error message on terminal

```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
Aborted (core dumped)
```

9. Please, describe detailedly what difficulty you are in

    I would like to spawn and remove Turtlebot models in Gazebo for a distributed simulation. However, when trying to delete a model using the Gazebo service, I get the error given above.

ROBOTIS-Will, 2019-12-07:
@ShiJbey 
Thanks for reporting. We'll take a look at the issue and get back to you.
Thank you.

ROBOTIS-Will, 2020-11-02:
This doesn't seem to be an issue of TurtleBot3, but the Gazebo.
Please feel free to reopen this when necessary.

closed 2020-11-02 by ROBOTIS-Will