ROBOTIS-GIT/turtlebot3 #522: Turtlebot3, change the velocity (linear and rotational) of the robot while driving depending on the traveled distance(angle) edit

Adel-kn, 2019-12-04:
-Robot: waffle Pi
-SBC: Raspberry Pi 3

OS installed in SBC: Raspbian
OS installed in Remote PC: Ubuntu 16.04 LTS (Xenial Xerus)
ROS version: 1.12.14

hello all, I would have a changeable velocity to drive the robot instead of the fixed one. I mean at the beginning of the movement I want the velocity to be low and then by increasing the traveled distance ethe velocity should increase until reaching the maximum value, then before reaching the goal, I want the velocity to start to decrease until '0' when arriving the goal. Any idea from you could help Thanks in advance

ROBOTIS-Will, 2019-12-04:
@Adel-kn 
If you want to control the acceleration / deceleration, you can also consider using Profile Acceleration of DYNAMIXEL instead of writing a velocity controlling code.
http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#profile-acceleration108

Adel-kn, 2019-12-05:
@ROBOTIS-Will 
Thanks for your help.
I read the information in the link you sent to me but actually I did not find how can I use the velocity profile to control the velocity based on the goal position

ROBOTIS-Will, 2019-12-12:
@Adel-kn The Profile Acceleration is saved in RAM area of DYNAMIXEL during power up and applied automatically for all Goal Position movement. Please take a loot at [this youtube video](https://youtu.be/gZWoyCvU-U8?t=68) that applies Profile Acceleration only for left to right motion of the wine glass. The orange juice in the glass represents the jerky / smooth motion for each direction.

closed 2020-01-22 by ROBOTIS-Will

ROBOTIS-Will, 2020-01-22:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.