ROBOTIS-GIT/turtlebot3 #523: how to get Turtlebot3 Omni code working

jennuine, 2019-12-04:
We have modified one of our turtlebot3s to be omnidirectional. I know that there is code from Arduino's Examples > turtlebot3 > turtlebot3_friends > turtlebot3_omni (http://emanual.robotis.com/docs/en/platform/turtlebot3/locomotion/#turtlebot3-friends-omni). Could you please explain how to properly load the script to the OpenCR board and how it works?

Thank you

ROBOTIS-Will, 2019-12-07:
@jennuine 
Please install [OpenCR board manager](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#porting-to-arduino-idewindows) from your Arduino IDE and upload the example.
You should be able to find the example after selecting OpenCR from **Tools > Board**, then go to **File > Examples > TurtleBot3 > turtlebot3_friends > turtlebot3_omni**.
Thank you.

jennuine, 2019-12-09:
Hi @ROBOTIS-Will, I have uploaded the example but it does not seem to be working with the rc100 remote control.

red24dog, 2019-12-10:
What is the github.com link to the OpenCR Turtlebot3_omni code?

jennuine, 2019-12-11:
@red24dog yes, that is the version I was trying to use.

ROBOTIS-Will, 2019-12-12:
@jennuine Could you give us more information on your hardware configuration? 
- Which DYNAMIXEL are you using and what are their ID and baudrates?
- What power source do you use?
- Which wireless interface do you use to connect RC100 to OpenCR?
- Which OpenCR port do you use to connect the wireless interface?

jennuine, 2019-12-19:
Hi @ROBOTIS-Will sorry for the delayed response. Given our limited time, we decided to proceed with a different configuration. Thank you for your help with this.

Regards,
Jenn

closed 2020-01-03 by ROBOTIS-Will