ROBOTIS-GIT/turtlebot3 #524: TurtleBot 3 Waffle Pi STL File

ChDillmann, 2019-12-30:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [X] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [X] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [X] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [X] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - HERE

8. Copy and Paste your error message on terminal

    - HERE

9. Please, describe detailedly what difficulty you are in

Hello everybody,
why do the stl files (waffle pi base and waffle base) show differences regarding the mounting (position of pcb support) of the laser scanner?
(see following pictures)
    
[image]

[image]

ROBOTIS-Will, 2020-01-03:
@ChDillmann 
Hi, the LIDAR is repositioned to attach OpenMANIPULATOR-X on TurtleBot3 Waffle Pi.

master-chrise, 2020-01-04:
@ROBOTIS-Will 
Thanks for your answer. It looks like the two file names (waffle_pi_base.stl and waffle_pi_base_for_open_manipulator.stl) on github have been swapped.

ChDillmann, 2020-01-05:
@ROBOTIS-Will
Thanks for your answer. It looks like the file names (waffle_pi_base.stl and waffle_pi_base_for_open_manipulator.stl) on github have been swapped.

ROBOTIS-Will, 2020-01-08:
@ChDillmann 
Thank you for reporting the issue. We'll fix the filename correctly.

PR #526 (2020-01-08): hotfix for #524
`waffle_pi_base.stl` and `waffle_pi_base_for_open_manipulator.stl` filenames are swapped.
Close #524
closed 2020-01-08 by ROBOTIS-Will