ROBOTIS-GIT/turtlebot3 #527: ERROR when i run bringup command on PC.

chenzhanjie, 2020-01-09:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - Burger

2. Which ROS is working with TurtleBot3?

    - ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Raspberry Pi 3

4. Which OS you installed in SBC?

    - Lubuntu

5. Which OS you installed in Remote PC?

    - Ubuntu 16.04 LTS (Xenial Xerus)

6. Write down software version and firmware version

    - Software version: ubuntu16.04 & Kinetic
    - Firmware version: N/A

7. Write down the commands you used in order

    - roslaunch turtlebot3_bringup turtlebot3_robot.launch

8. Copy and Paste your error message on terminal

    - [ERROR] [1578540338.207788006]: An exception was thrown: open: No such file or directory
[INFO] [1578540338.441196]: ROS Serial Python Node
[INFO] [1578540338.443938]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1578540338.444740]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

9. Please, describe detailedly what difficulty you are in

    - I want to start turtlebot3 using commands on my PC, but when i run the command:
roslaunch turtlebot3_bringup turtlebot3_robot.launch
I received the error information:
[ERROR] [1578540338.207788006]: An exception was thrown: open: No such file or directory
[turtlebot3_lds-2] process has died [pid 23707, exit code 255, cmd /opt/ros/kinetic/lib/hls_lfcd_lds_driver/hlds_laser_publisher __name:=turtlebot3_lds __log:=/home/gree/.ros/log/a4db5e00-328f-11ea-9652-00d86193d908/turtlebot3_lds-2.log].
log file: /home/gree/.ros/log/a4db5e00-328f-11ea-9652-00d86193d908/turtlebot3_lds-2*.log
[INFO] [1578540338.441196]: ROS Serial Python Node
[INFO] [1578540338.443938]: Connecting to /dev/ttyACM0 at 115200 baud
[ERROR] [1578540338.444740]: Error opening serial: [Errno 2] could not open port /dev/ttyACM0: [Errno 2] No such file or directory: '/dev/ttyACM0'

honestly, I don't have  idea about it, so please help me explain why the error occurred. i guess that PC cannot detect the inform of USB of the turtlebot through ROS remote control, of course it is just a guess.  
and how can i fix this problem?

ROBOTIS-Will, 2020-01-14:
@chenzhanjie 
Hi, it is recommended to use Raspbian for Raspberry Pi on TurtleBot3.
Please refer to below eManual instruction to download the Raspbian image.
http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian
Thank you.

chenzhanjie, 2020-01-14:
@ROBOTIS-Will Ok，than you.

closed 2020-01-22 by ROBOTIS-Will