ROBOTIS-GIT/turtlebot3 #528: turtlebot doesn't move

mjmanny, 2020-01-16:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - ros2 run turtlebot3_teleop teleop_keyboard

8. Copy and Paste your error message on terminal

    - I don't have any message error, but it doesn't work.

9. Please, describe detailedly what difficulty you are in

    - I followed all the steps in the emanual, and it was all fine, but the robot doesn't move.

    - I tried to use RC-100 but the robot doesn't.

mjmanny, 2020-01-19:
Finally!
My TB3 works, it was just that one of  the dynamixel cables was in correct place.

Thank's

closed 2020-01-19 by mjmanny

ishaanmht, 2020-07-15:
How did you resolve this ?