ROBOTIS-GIT/turtlebot3 #529: SImulation failing in ROS Melodic, Gazebo9

pushkalkatara, 2020-01-17:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have? (Gazebo Simulation)

    - [x] Burger
    - [x] Waffle
    - [x] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (Simulation Gazebo)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - roslaunch turtlebot3_gazebo turtlebot3_world.launch

8. Copy and Paste your error message on terminal

```bash
[rospy.client][INFO] 2020-01-17 16:52:52,694: init_node, name[/spawn_urdf], pid[20632]
[xmlrpc][INFO] 2020-01-17 16:52:52,694: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2020-01-17 16:52:52,695: Started XML-RPC server [http://pushkalkatara:45779/]
[rospy.init][INFO] 2020-01-17 16:52:52,695: ROS Slave URI: [http://pushkalkatara:45779/]
[rospy.impl.masterslave][INFO] 2020-01-17 16:52:52,695: _ready: http://pushkalkatara:45779/
[xmlrpc][INFO] 2020-01-17 16:52:52,696: xml rpc node: starting XML-RPC server
[rospy.registration][INFO] 2020-01-17 16:52:52,697: Registering with master node http://localhost:11311
[rospy.init][INFO] 2020-01-17 16:52:52,796: registered with master
[rospy.rosout][INFO] 2020-01-17 16:52:52,796: initializing /rosout core topic
[rospy.rosout][INFO] 2020-01-17 16:52:52,799: connected to core topic /rosout
[rospy.simtime][INFO] 2020-01-17 16:52:52,802: initializing /clock core topic
[rospy.simtime][INFO] 2020-01-17 16:52:52,811: connected to core topic /clock
[rospy.internal][INFO] 2020-01-17 16:52:52,818: topic[/clock] adding connection to [http://pushkalkatara:44633/], count 0
[rosout][INFO] 2020-01-17 16:52:52,826: Loading model XML from ros parameter robot_description
[rosout][INFO] 2020-01-17 16:52:52,855: Waiting for service /gazebo/spawn_urdf_model
[rosout][INFO] 2020-01-17 16:52:52,865: Calling service /gazebo/spawn_urdf_model
[rospy.internal][INFO] 2020-01-17 16:52:53,067: topic[/rosout] adding connection to [/rosout], count 0
[rosout][INFO] 2020-01-17 16:52:53,876: Spawn status: SpawnModel: Successfully spawned entity
[rospy.core][INFO] 2020-01-17 16:52:53,877: signal_shutdown [atexit]
[rospy.internal][INFO] 2020-01-17 16:52:53,879: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2020-01-17 16:52:53,880: topic[/clock] removing connection to http://pushkalkatara:44633/
[rospy.internal][WARNING] 2020-01-17 16:52:53,880: Unknown error initiating TCP/IP socket to pushkalkatara:45583 (http://pushkalkatara:44633/): Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 558, in connect
    self.read_header()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 653, in read_header
    self._validate_header(read_ros_handshake_header(sock, self.read_buff, self.protocol.buff_size))
AttributeError: 'NoneType' object has no attribute 'buff_size'

[rospy.topics][ERROR] 2020-01-17 16:52:53,881: Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 326, in close
    c.close()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 841, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

[rospy.impl.masterslave][INFO] 2020-01-17 16:52:53,881: atexit
```

9. Please, describe detailedly what difficulty you are in

    - Failing to do a simulation in Gazebo9.

ROBOTIS-Will, 2020-01-22:
@pushkalkatara 
I've tried with your setup on Linux Mint(Bionic) + ROS Melodic + Gazebo9.11.0 + python 2.7.17 and couldn't reproduce your issue.
If you have downloaded binary TurtleBot3 package using `apt-get` command, please try with source code download as written in the eManual.

pushkalkatara, 2020-01-22:
I figured out that I missed installing ros-gazebo dependencies. Installing them using apt solved the issues. Thanks for the response. Closing the issue.

closed 2020-01-22 by pushkalkatara

nrkumar93, 2020-03-12:
What ros-gazebo dependencies did you isntall?

uchaudh, 2020-08-13:
I had the same issue but resolved it by installing the developer package.
`sudo apt install libgazebo9-dev`