ROBOTIS-GIT/turtlebot3 #530: Issue in building turtlebot3 packages with ROS2 on PC

pankhurivanjani, 2020-01-22:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?(SImulation)

    - [x ] Burger
    - [x ] Waffle
    - [x ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - colcon build --symlink-install

8. Copy and Paste your error message on terminal

    -
```
pankhuri@pankhuri-iMac:~/turtlebot3_ws$ colcon build --symlink-install
Starting >>> turtlebot3_msgs
Starting >>> turtlebot3_description
Starting >>> dynamixel_sdk
Starting >>> hls_lfcd_lds_driver
Finished <<< turtlebot3_description [0.39s]                              
Starting >>> turtlebot3_cartographer
Finished <<< hls_lfcd_lds_driver [0.49s]                                 
Starting >>> turtlebot3_navigation2
Finished <<< dynamixel_sdk [0.83s]                                       
Starting >>> turtlebot3_teleop
--- stderr: turtlebot3_msgs                                              
In file included from /home/pankhuri/turtlebot3_ws/build/turtlebot3_msgs/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/dds_opensplice/sound__type_support.cpp:19:0:
/home/pankhuri/turtlebot3_ws/build/turtlebot3_msgs/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/sound__rosidl_typesupport_opensplice_cpp.hpp:41:25: error: ‘turtlebot3_msgs::srv::dds_’ has not been declared
   turtlebot3_msgs::srv::dds_::Sound_Request_ & dds_message);
                         ^~~~
/home/pankhuri/turtlebot3_ws/build/turtlebot3_msgs/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/sound__rosidl_typesupport_opensplice_cpp.hpp:41:46: error: expected ‘,’ or ‘...’ before ‘&’ token
   turtlebot3_msgs::srv::dds_::Sound_Request_ & dds_message);
                                              ^
/home/pankhuri/turtlebot3_ws/build/turtlebot3_msgs/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/sound__rosidl_typesupport_opensplice_cpp.hpp:50:31: error: ‘dds_’ in namespace ‘turtlebot3_msgs::srv’ does not name a type
[... 1258 lines cut ...]
/home/pankhuri/turtlebot3_ws/build/turtlebot3_msgs/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/dds_opensplice/sound__type_support.cpp:1555:55: error: expected primary-expression before ‘>’ token
   auto typed_requester = reinterpret_cast<RequesterT *>(requester);
                                                       ^
/home/pankhuri/turtlebot3_ws/build/turtlebot3_msgs/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/dds_opensplice/sound__type_support.cpp:1555:67: error: expected ‘)’ before ‘;’ token
   auto typed_requester = reinterpret_cast<RequesterT *>(requester);
                                                                   ^
make[2]: *** [CMakeFiles/turtlebot3_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/turtlebot3_msgs/srv/dds_opensplice/sound__type_support.cpp.o] Error 1
make[1]: *** [CMakeFiles/turtlebot3_msgs__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< turtlebot3_msgs	[ Exited with code 2 ]
Finished <<< turtlebot3_cartographer [3.55s]     
Aborted  <<< turtlebot3_teleop
Aborted  <<< turtlebot3_navigation2

Summary: 4 packages finished [4.21s]
  1 package failed: turtlebot3_msgs
  2 packages aborted: turtlebot3_navigation2 turtlebot3_teleop
  1 package had stderr output: turtlebot3_msgs
  6 packages not processed
```

9. Please, describe detailedly what difficulty you are in

    - I am trying to build turtlebot3 with ros2 to simulate in gazebo. I am following the tutorial provided here [turtlebot3-ros2_tutorial](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/), instructions from PC setup and then simulations. I am getting error in building up the turtlebot2 packages with colcon build.  It seems some issue with opensplice with ros-dashing. I have used the versions provided on emanual tutorial.

ROBOTIS-Will, 2020-01-23:
@pankhurivanjani 
If you have installed the opensplice, try building the source after uninstalling the opensplice.

pankhurivanjani, 2020-01-23:
That solved the issue. 
Thank you!

closed 2020-01-23 by pankhurivanjani