ROBOTIS-GIT/turtlebot3 #531: TB3-Burger: RViz2 received SIGABRT(tb3_cartographer.rviz)

KangDroid, 2020-01-23:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [x] etc (ROS2 Master branch - self-compiled version)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x] etc (NVidia Jetson Nano)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [x] etc (NVidia Jetson Nano default OS provided from NVidia - [which is custom ubuntu 18.04])

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
    - Cannot confirm. But, I have installed & uploaded firmware with <br> https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS2/latest/opencr_update.tar.bz2 

7. Write down the commands you used in order
    - Basically I followed instructions from <br> http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/ to <br> http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_slam/

```
# The main one caused trouble;
$ ros2 launch turtlebot3_cartographer cartographer.launch.py
```

8. Copy and Paste your error message on terminal

- When I first launched with <br>
ros2 launch turtlebot3_cartographer cartographer.launch.py
```
[rviz2-3] [INFO] [1579740882.098278078] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1579740882.098691686] [rviz2]: OpenGl version: 3.3 (GLSL 3.3)
[rviz2-3] [INFO] [1579740882.145527862] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] ../src/gallium/auxiliary/tgsi/tgsi_parse.c:268:tgsi_parse_token: Assertion `0' failed.
[ERROR] [rviz2-3]: process has died [pid 37459, exit code -6, cmd '/home/kangdroid/ros2_foxy/install/rviz2/lib/rviz2/rviz2 -d /home/kangdroid/custom_packages/install/turtlebot3_cartographer/share/turtlebot3_cartographer/rviz/tb3_cartographer.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_cpyh1tlk'].
```

- When I separately launch rviz2 with <br>
```
$ /home/kangdroid/ros2_foxy/install/rviz2/lib/rviz2/rviz2 -d /home/kangdroid/custom_packages/install/turtlebot3_cartographer/share/turtlebot3_cartographer/rviz/tb3_cartographer.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_cpyh1tlk
```
while other launch nodes are running from command on 8.<br>
(since rviz2 is killed when executing command on 8, and not giving exact information)
```
[INFO] [1579741233.584949039] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1579741233.586029910] [rviz2]: OpenGl version: 3.3 (GLSL 3.3)
[INFO] [1579741233.645795310] [rviz2]: Stereo is NOT SUPPORTED
../src/gallium/auxiliary/tgsi/tgsi_parse.c:91:next_token: Assertion `!tgsi_parse_end_of_tokens( ctx )' failed.
Aborted (core dumped)
```
8-1. A Workround

 - Modifying turtlebot3_cartographer/rviz/tb3_cartographer.rviz<br>
from e97b2abde783ca56665c4332b29db4d90f1f9e6a to 6e814638b941cc6f1d7ebcd2cd178031435497f7 does work very well(Except we need to set /map topic on rviz - but that isn't a problem at all.)

- More precisely, in commit e97b2abde783ca56665c4332b29db4d90f1f9e6a ,
https://github.com/ROBOTIS-GIT/turtlebot3/commit/e97b2abde783ca56665c4332b29db4d90f1f9e6a#diff-d0b5d64586de4ed66a4b98407a61abbdR269 
to
https://github.com/ROBOTIS-GIT/turtlebot3/commit/e97b2abde783ca56665c4332b29db4d90f1f9e6a#diff-d0b5d64586de4ed66a4b98407a61abbdR329 
causes those error described in 8 (line 269~ line 329)

9. Please, describe detailedly what difficulty you are in

    - Cannot launch SLAM nodes(followed http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_slam/#run-slam-nodes

KangDroid, 2020-01-28:
Any updates or author comments?

rjshim, 2020-01-31:
@KangDroid 

Are you testing the cartographer package with the ROS2 Dashing version? If you are testing it by the latest version, it might have compatibility issues.

KangDroid, 2020-01-31:
@rjshim 

As I can remember, I didn't installed cartographer package via apt. Rather installed it from source.
Speaking of compatibility issue, https://github.com/ros2/cartographer_ros still on dashing and isn't updated since December, 2019. Is that just low-priority migrating ROS1 to ROS2?

For now, as I mentioned Pull Request(https://github.com/ROBOTIS-GIT/turtlebot3/pull/532) removing cartographer-related lines will fix issue temporarily.

I will double-check whole ros library version and give you more information about it. 
I am week-off from laboratory, so I will re-mention you when I get detailed information.

Thanks.

JaehyunShim, 2020-10-16:
I will close this thread as there have been no recent updates.

closed 2020-10-16 by JaehyunShim