ROBOTIS-GIT/turtlebot3 #533: Raspberry Pi camera installation process not working

jamescoady, 2020-01-28:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

     - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

I am unsure how to find this?

7. Write down the commands you used in order

    - sudo raspi-config
    I then selected interfacing options and then P1 camera. I then enabled the camera interface and reboot the system as per instructions (link below).
http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/ 

   -raspistill -v -o test.jpg
  This was to test the system.

8. Copy and Paste your error message on terminal

mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: failed to create camera component
mmal: main: failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

9. Please, describe detailedly what difficulty you are in

    - I am a new beginner to this and am currently completing the turtlebot e-manual.

ROBOTIS-Will, 2020-01-31:
@jamescoady 
It seems this can be caused by faulty camera connection.
Check all your ribbon cable is properly fixed to the port and firmly locked with the locker.

jamescoady, 2020-01-31:
> @jamescoady
> It seems this can be caused by faulty camera connection.
> Check all your ribbon cable is properly fixed to the port and firmly locked with the locker.
> https://www.raspberrypi.org/forums/viewtopic.php?t=39996

This was the issue, my bad, thank you :)

closed 2020-01-31 by jamescoady