ROBOTIS-GIT/turtlebot3 #534: TurtleBot3 tutorial, telekeyboard, robot doesn't move

dwindy, 2020-01-28:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [1 ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [1 ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [1 ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [1 ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [1 ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [1.12.14]

7. Write down the commands you used in order

    - Remote-PC
      roscore
    - Turtlebot
      roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - Remote-PC
      export TURTLEBOT3_MODEL=buerger
      rosrun rviz rviz -d rospack find turtlebot3_description /rviz/model.rviz
      rqt
      roslaunch turtlstrong textebot3_teleop turtlebot3_teleop_key.launch

8. Copy and Paste your error message on terminal
     CTRL-C to quit

     currently:	linear vel 0.01	 angular vel 0.0 
     currently:	linear vel 0.02	 angular vel 0.0 
     currently:	linear vel 0.03	 angular vel 0.0 
     currently:	linear vel 0.04	 angular vel 0.0 

9. Please, describe detailedly what difficulty you are in

    - The robot doesn't move.
      In rqt, the topic "cmd_vel" is updating the linear speed.
     but the topic "cmd_vel_rc100" is keep linear speed 0.

THANK YOU

dwindy, 2020-01-28:
I run echo 'cmd_vel' on the robot terminal.
it receive the speed info from Remote-PC.

ROBOTIS-Will, 2020-01-31:
@dwindy 
Before you run the rviz command, please run below command from [eManual](http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#load-a-turtlebot3-on-rviz).
```code
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
```

dwindy, 2020-02-04:
> @dwindy
> Before you run the rviz command, please run below command on your Remote PC as written in [eManual](http://emanual.robotis.com/docs/en/platform/turtlebot3/bringup/#load-a-turtlebot3-on-rviz).
> 
> ```
> $ roslaunch turtlebot3_bringup turtlebot3_remote.launch
> ```

Thanks for the help.
I test with some commands.
Finally found out that the Turtblebot3 connected to a wireless network while the virtual machine didn't.

closed 2020-02-04 by dwindy