ROBOTIS-GIT/turtlebot3 #535: Turtlebot3

tahir1069, 2020-01-30:
1. Which TurtleBot3 you have?

    - Waffle Pi

2. Which ROS is working with TurtleBot3?

    - ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - Raspberry Pi 3

4. Which OS you installed in SBC?

    - Raspbian

5. Which OS you installed in Remote PC?

    - Ubuntu 18.04 LTS (Bionic Beaver)

I have a turtlebot3 Waffle Pi with the above mentioned configuration. Currently I have four motors and hence four wheels. I want to make my robot omni-directional with four meccanum wheels. I have seen and uploaded the example code provided in Arduino to OpenCR board which actually didn't worked at all. So I start looking into what's the difference between in TB3 core and the example provided. I found that all ROS related stuff is done inside TB3 Core code which actually is not present in  TB3 meccanum example. 

I have found an example [here](https://github.com/Seunghooon/turtlebot3_mecanum). Which actually is not working as an issue raised [here](https://github.com/Seunghooon/turtlebot3_mecanum/issues/3).

I am looking for any leads or suggestions for the project. Is there anything available already so that I do not need to reinvent the wheel. Or if I have to how and what are the things I need to develop or consider.

ROBOTIS-Will, 2020-01-31:
@tahir1069 
The Locomotion examples are only to present different types of locomotion based on TurtleBot3 platform.
As those example code is written for remote control with ROBOTIS RC-100 controller, it is not compatible with ROS.

ROBOTIS-Will, 2020-02-28:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if necessary. Thanks.

closed 2020-02-28 by ROBOTIS-Will