ROBOTIS-GIT/turtlebot3 #536: Configuring tf tree for multiple turtlebot navigation in real environment

randyd55, 2020-01-30:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [X] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [X] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - [turtlebot0] ROS_NAMESPACE=tb3_0 roslaunch turtlebot3_bringup turtlebot3_robot.launch
      [turtlebot1] ROS_NAMESPACE=tb3_1 roslaunch turtlebot3_bringup turtlebot3_robot.launch
      [Remote] ROS_NAMESPACE=tb3_0 roslaunch turtlebot3_navigation turtlebot3_navigation_0.launch map_file:=$HOME/conference.yaml set_base_frame:=tb3_0/base_footprint set_odom_frame:=tb3_0/odom set_map_frame:=tb3_0/map
       [Remote] ROS_NAMESPACE=tb3_1 roslaunch turtlebot3_navigation turtlebot3_navigation_1.launch map_file:=$HOME/conference.yaml set_base_frame:=tb3_1/base_footprint set_odom_frame:=tb3_1/odom set_map_frame:=tb3_1/map

8. Copy and Paste your error message on terminal

    - HERE

9. Please, describe detailedly what difficulty you are in

    - I am trying to run multiple turtlebot3 navigation in a real life environment using the instruction for a single turtlebot found here http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#run-navigation-nodes. My issue is configuring the tf trees for each turtlebot so they do not overlap. When I run the above commands, no errors are printed but the position of each robot in rviz oscillates between the two robots' positions because their tf_trees overlap:

[image]

I am using edited versions of the turtlebot3_navigation.rviz configuration file to configure the topics used by each turtlebot. I can post the full text of each file but basically I created 2 new rviz config files turtlebot3_navigation_0.rviz and turtlebot3_navigation_1.rviz and appended tb3_0/ and tb3_1/ to the beginning of the topics used in the file.

ROBOTIS-Will, 2020-01-31:
@randyd55 
Please refer to below eManual for loading multiple TurtleBot3
http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#load-multiple-turtlebot3s

ROBOTIS-Will, 2020-01-31:
I just realized that you had the same issue before that is not closed yet.
I'm going to close this thread and keep working on #517 thread.
Thank you.

closed 2020-01-31 by ROBOTIS-Will