ROBOTIS-GIT/turtlebot3 #537: Issue when adding repository package for ROS node for the Raspberry Pi Camera Module

jamescoady, 2020-01-31:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

     - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
  
3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3

4. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later
  
5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

  -Adding the cource to my apt lists:
sudo sh -c 'echo "deb https://packages.ubiquityrobotics.com/ubuntu/ubiquity xenial main" > /etc/apt/sources.list.d/ubiquity-latest.list'

 -Add signing key to trusted list:
sudo apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key C3032ED8

 -sudo apt-get update

 -sudo apt install ros-kinetic-raspicam-node

8. Copy and Paste your error message on terminal

 Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-raspicam-node

9. Please, describe detailedly what difficulty you are in

 I am new to linux, ubuntu etc. I am working through the turtlebot e-manual and am on the pano demo stage (see link)
http://emanual.robotis.com/docs/en/platform/turtlebot3/applications/#applications

Panorama demo requires to install raspicam_node package. Instructions for installing this package can be found at Gihub Link (see link)
https://github.com/UbiquityRobotics/raspicam_node .

From here I am trying to add the repository through this link
sudo apt install ros-kinetic-raspicam-node

Once I have followed all the steps I try to run:
sudo apt install ros-kinetic-raspicam-node

I then get the error shown in the error section.

ROBOTIS-Will, 2020-02-10:
@jamescoady 
Raspberry Pi 3 does not fully support Ubunut 16.04.
In order to run examples in ROBOTIS eManual, please download and install [the Raspbian image](http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#install-linux-based-on-raspbian)
Thank you.

ROBOTIS-Will, 2020-02-28:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if necessary. Thanks.

closed 2020-02-28 by ROBOTIS-Will