ROBOTIS-GIT/turtlebot3 #538: Turtlebot battery monitoring

AGmochi, 2020-02-14:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [x] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]
I am not sure where I would find this info.
7. Write down the commands you used in order

    roscore
   roslaunch turtlebot3_bringup turtlebot3_core.launch
  rqt

8. Copy and Paste your error message on terminal

    not really an error

9. Please, describe detailedly what difficulty you are in

    Hi, I would mainly like to confirm on how the voltage is being read. Is it because due to the fact that OpenCR could be treated like an arduino, there is a specific function in the code that is reading the voltage or is there an internal voltage sensor?
In addition, is it possible to change the provided code so that current can be read and recorded by the turtlebot or the remote PC? Or would a seperate current sensor be required in order to read the current?

Thank you.

ROBOTIS-Will, 2020-02-19:
@AGmochi 
Hi, the battery voltage is read from OpenCR and delivered to RPi via rosserial. You can read the topic `/battery_state` to check the voltage.
Publishing the battery status code is written in [Arduino code here](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L301).
Unfortunately, OpenCR does not have current sensing feature, but by evaluating the voltage level, you can assume the power consumption during certain period of time.
Thank you.

AGmochi, 2020-02-19:
Hi, 

Then is it possible to add a separate current sensor to the OpenCR to let it read current?

Thank you.

ROBOTIS-Will, 2020-02-28:
@AGmochi 
Yes, you can add additional sensors on OpenCR as there are multiple GPIO and ADC as well as serial ports available.
[http://emanual.robotis.com/docs/en/parts/controller/opencr10/#layoutpin-map](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#layoutpin-map)

AGmochi, 2020-03-04:
Hi
I am slightly confused in relation to the coding. I have a current sensor connected to the OpenCR, so if I want to let ros read and save the current data, does that mean I am editing the core code for the OpenCR through the Arduino IDE? Or is there another method?

ROBOTIS-Will, 2020-03-13:
@AGmochi 
If you connected the sensor output to the OpenCR input pin, you should modify the Arduino code so that the current reading from OpenCR can be sent to the RPi.

ROBOTIS-Will, 2020-04-21:
This issue will be closed as there was no activity on this thread. Please feel free to reopen the issue whenever needed.

closed 2020-04-21 by ROBOTIS-Will