ROBOTIS-GIT/turtlebot3 #539: simulation with cartographer not working. Check failed: sensor_to_tracking->translation().norm() < 1e-5

dwindy, 2020-02-14:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [1.12.14]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

`roslaunch turtlebot3_gazebo turtlebot3_world.launch`

`roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=cartographer`

8. Copy and Paste your error message on terminal

> [FATAL] [1581699215.451365267, 8.974000000]: F0214 16:53:35.000000 41313 sensor_bridge.cc:136] Check failed: sensor_to_tracking->translation().norm() < 1e-5 The IMU frame must be colocated with the tracking frame. Transforming linear acceleration into the tracking frame will otherwise be imprecise.

> [cartographer_node-3] process has died [pid 41313, exit code -6, cmd /home/bingxin/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/bingxin/catkin_ws/install_isolated/share/turtlebot3_slam/config -configuration_basename turtlebot3_lds_2d.lua /imu:=/flat_imu __name:=cartographer_node __log:=/home/bingxin/.ros/log/825f9288-4f4a-11ea-9898-000c29f89067/cartographer_node-3.log].
log file: /home/bingxin/.ros/log/825f9288-4f4a-11ea-9898-000c29f89067/cartographer_node-3*.log

9. Please, describe detailedly what difficulty you are in

I followed the turtlebot e-manual.
successfully run the turtlebot with cartographer in real environment.
but met the fatal error with simulation environment. 
btw, simulation with gammping works.

any ideas?
Thanks alot

ROBOTIS-Will, 2020-02-19:
@dwindy 
Hi,
Did you install the Cartographer 0.2.0 as described in the [eManual](http://emanual.robotis.com/docs/en/platform/turtlebot3/slam/#run-slam-nodes)?
If you have installed with binary packages(such using as `apt-get` command), it is likely that 0.3.0 was installed.
Thank you.

dwindy, 2020-02-19:
@ROBOTIS-Will 
Hi,
Thanks for replying.
I redo the installation again and confirmed followed the eManual.
(git clone, src the protobuf3.sh, rosdep, catkin_make_isolated, source the bash )
Still got the FATAL problem.
Is there anyway to check the cartopraher version number?
Thank you.

dwindy, 2020-02-19:
[image]
[image]
I uploaded the node graph in case it is helpful

dwindy, 2020-02-27:
solve.
I git clone the cartographer_turtlebot.
managed to modify the launch and lua files to fit my bag.
during that, I coincidently meet the same problem of 

> "Check failed: sensor_to_tracking->translation().norm() < 1e-5".

Which inspired me to check the 

> turtlebot3_slam.launch 

the launch file leads to another lua file

> turtlebot3_lds_2d.lua
[image]

I am not sure whether the comment said it's a bug is right or not.
But followed by the inspiration of cartographer_turtlebot_demo.bag,
the tracking_frame = value should be equal to the imu topic frame.
In the demo.bag, it is "gryo_link" frame, in my simulation gazebo environment, it is "base_footprint".

So, change that, it works.

closed 2020-02-27 by dwindy