ROBOTIS-GIT/turtlebot3 #542: ROS1 Release with the python 3 fix.

bionade24, 2020-02-24:
Hello, I'm one of the ROS melodic AUR maintainers. Please make a release with @acxz  python3 patch. If you don't want to do this, please tell us, so that we can patch it.

acxz, 2020-02-24:
I think this is the specific commit: https://github.com/ROBOTIS-GIT/turtlebot3/commit/ddca617172a3549a3077682cd2cf7dc377616e01

ROBOTIS-Will, 2020-02-28:
@bionade24 @acxz 
Hi, it had been updated to the most recent ROS1 master branch only, and had not been applied to Melodic branch yet. 
I'll apply this commit on Melodic.
Thank you.

closed 2020-03-23 by ROBOTIS-Will

bionade24, 2020-03-23:
We still don't have a release. When will it come?

ROBOTIS-Will, 2020-03-26:
@bionade24 
Hi, the change has been applied to melodic branch, so please allow some time to be applied in the binary package.
You can also clone the source and use melodic-devel branch.

bionade24, 2020-03-26:
We are already patching the files during the build process,

PR #550 (2020-03-23): Latest update
close #542