ROBOTIS-GIT/turtlebot3 #545: why is local costmap empty when launching move_base with namespace in turtlebot3_house?

mariocj, 2020-03-10:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [1.0.0]
    - Firmware version: [1.2.0]

7. Write down the commands you used in order
 
Hello, 
     I am beginner to ROS, based on tutorial [Virtual SLAM by Multiple TurtleBot3s](http://emanual.robotis.com/docs/en/platform/turtlebot3/simulation/#virtual-slam-by-multiple-turtlebot3s) I could successfully run all three turtlebots, however, for the past few weeks, i am trying to launch move_base for each turtlebot so every turtlebot can explore the map using explore_lite package. I reconfigured move_base for each turtlebot and basically everything works except local_costmap which is empty and returns all zeros so there is no obstacle avoidance. I also get a warn messages but still cannot figure out what i am doing wrong.

8. Copy and Paste your error message on terminal

`[ WARN] [1583252595.618832497, 572.601000000]: MessageFilter [target=map base_scan ]: Dropped 100.00% of messages so far. Please turn the [ros.costmap_2d.message_notifier] rosconsole logger to DEBUG for more information.`
`[ WARN] [1583252595.619562044, 572.601000000]: MessageFilter [target=tb3_0/odom base_scan ]: Dropped 100.00% of messages so far. Please turn the [ros.costmap_2d.message_notifier] rosconsole logger to DEBUG for more information.`
`[DEBUG] [1583575121.766547302, 824.820000000]: MessageFilter [target=map base_scan ]: Removed oldest message because buffer is full, count now 50 (frame_id=tb3_0/base_scan, stamp=814.806000)`
`[DEBUG] [1583575121.766656904, 824.820000000]: MessageFilter [target=map base_scan ]: Added message in frame tb3_0/base_scan at time 824.807, count now 50`
`[DEBUG] [1583575121.766870426, 824.820000000]: MessageFilter [target=tb3_0/odom base_scan ]: Removed oldest message because buffer is full, count now 50 (frame_id=tb3_0/base_scan, stamp=814.806000)`
`[DEBUG] [1583575121.767069814, 824.822000000]: MessageFilter [target=tb3_0/odom base_scan ]: Added message in frame tb3_0/base_scan at time 824.807, count now 50`

9. Please, describe detailedly what difficulty you are in

    **Down below is my move_base for tb3_0. I did not change other param files, however i have tried to change sensor frame: base_scan and scan topic in costmap_common_params_burger.yaml, but the local_costmap was still empty.** 

Thank you for any help.

```
<launch>
  <!-- Arguments -->
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="cmd_vel_topic" default="tb3_0/cmd_vel" />
  <arg name="odom_topic" default="tb3_0/odom" />
  <arg name="move_forward_only" default="true"/>

  <!-- move_base -->
  <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
    <param name="base_local_planner" value="dwa_local_planner/DWAPlannerROS" />
    <rosparam file="$(find turtlebot3_navigation)/param/costmap_common_params_$(arg model).yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find turtlebot3_navigation)/param/costmap_common_params_$(arg model).yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find turtlebot3_navigation)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find turtlebot3_navigation)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find turtlebot3_navigation)/param/move_base_params.yaml" command="load" />
    <rosparam file="$(find turtlebot3_navigation)/param/dwa_local_planner_params_$(arg model).yaml" command="load" />
    <param name="global_costmap/robot_base_frame" value="tb3_0/base_footprint"/>
    <param name="global_costmap/global_frame" value="tb3_0/map"/> 
    <param name="local_costmap/robot_base_frame" value="tb3_0/base_footprint"/>
    <param name="local_costmap/global_frame" value="tb3_0/odom"/>

    <param name="DWAPlannerROS/min_vel_x" value="0.0" if="$(arg move_forward_only)" />
  </node>
</launch>
```

Here is my  [rqt_tf_tree](https://imgur.com/cZaJL03) output:

ROBOTIS-Will, 2020-03-20:
@mariocj 
I'm not sure if you are trying to run a Navigation with multiple TurtleBot3 in simulation.
I'd create separate namespaces for each TurtleBot3 when running the move_base node so each of them can have different costmaps.

mariocj, 2020-03-21:
@ROBOTIS-Will  

Actually, i am not trying to run navigation with multiple turtlebots. I am trying to run exploration with multiple turtlebots using [explore_lite](http://wiki.ros.org/explore_lite) package, which is working for turtlebot 3 out of the box. But to run this package properly, i need to have a properly configured move_base node for different turtlebots, that is why i posted here my move_base for tb3_0, which is working fine, but does not create local costmap. 

This is my rostopic list, after i launch all commands from tutorial 11.2.1.6 and then move_base node with command: ROS_NAMESPACE=tb3_0 roslaunch move_base move_base_tb3_0.launch (trying just one turtlebot with namespace at the time):

rostopic list:

/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/gazebo_gui/parameter_descriptions
/gazebo_gui/parameter_updates
/map
/rosout
/rosout_agg
/tb3_0/cmd_vel
/tb3_0/explore/frontiers
/tb3_0/imu
/tb3_0/joint_states
/tb3_0/map
/tb3_0/map_metadata
/tb3_0/map_updates
/tb3_0/move_base/DWAPlannerROS/cost_cloud
/tb3_0/move_base/DWAPlannerROS/global_plan
/tb3_0/move_base/DWAPlannerROS/local_plan
/tb3_0/move_base/DWAPlannerROS/parameter_descriptions
/tb3_0/move_base/DWAPlannerROS/parameter_updates
/tb3_0/move_base/DWAPlannerROS/trajectory_cloud
/tb3_0/move_base/NavfnROS/plan
/tb3_0/move_base/cancel
/tb3_0/move_base/current_goal
/tb3_0/move_base/feedback
/tb3_0/move_base/global_costmap/costmap
/tb3_0/move_base/global_costmap/costmap_updates
/tb3_0/move_base/global_costmap/footprint
/tb3_0/move_base/global_costmap/inflation_layer/parameter_descriptions
/tb3_0/move_base/global_costmap/inflation_layer/parameter_updates
/tb3_0/move_base/global_costmap/obstacle_layer/parameter_descriptions
/tb3_0/move_base/global_costmap/obstacle_layer/parameter_updates
/tb3_0/move_base/global_costmap/parameter_descriptions
/tb3_0/move_base/global_costmap/parameter_updates
/tb3_0/move_base/global_costmap/static_layer/parameter_descriptions
/tb3_0/move_base/global_costmap/static_layer/parameter_updates
/tb3_0/move_base/goal
/tb3_0/move_base/local_costmap/costmap
/tb3_0/move_base/local_costmap/costmap_updates
/tb3_0/move_base/local_costmap/footprint 
/tb3_0/move_base/local_costmap/inflation_layer/parameter_descriptions
/tb3_0/move_base/local_costmap/inflation_layer/parameter_updates
/tb3_0/move_base/local_costmap/obstacle_layer/parameter_descriptions
/tb3_0/move_base/local_costmap/obstacle_layer/parameter_updates
/tb3_0/move_base/local_costmap/parameter_descriptions
/tb3_0/move_base/local_costmap/parameter_updates
/tb3_0/move_base/parameter_descriptions
/tb3_0/move_base/parameter_updates
/tb3_0/move_base/result
/tb3_0/move_base/status
/tb3_0/move_base_simple/goal
/tb3_0/odom
/tb3_0/scan
/tb3_0/turtlebot3_slam_gmapping/entropy
/tb3_1/camera/depth/camera_info
/tb3_1/camera/depth/image_raw
/tb3_1/camera/depth/points
/tb3_1/camera/parameter_descriptions
/tb3_1/camera/parameter_updates
/tb3_1/camera/rgb/camera_info
/tb3_1/camera/rgb/image_raw
/tb3_1/camera/rgb/image_raw/compressed
/tb3_1/camera/rgb/image_raw/compressed/parameter_descriptions
/tb3_1/camera/rgb/image_raw/compressed/parameter_updates 
/tb3_1/camera/rgb/image_raw/compressedDepth
/tb3_1/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/tb3_1/camera/rgb/image_raw/compressedDepth/parameter_updates
/tb3_1/camera/rgb/image_raw/theora
/tb3_1/camera/rgb/image_raw/theora/parameter_descriptions
/tb3_1/camera/rgb/image_raw/theora/parameter_updates
/tb3_1/cmd_vel
/tb3_1/explore/frontiers
/tb3_1/imu
/tb3_1/joint_states
/tb3_1/map
/tb3_1/map_metadata
/tb3_1/map_updates
/tb3_1/move_base/DWAPlannerROS/cost_cloud
/tb3_1/move_base/DWAPlannerROS/global_plan
/tb3_1/move_base/DWAPlannerROS/local_plan
/tb3_1/move_base/DWAPlannerROS/parameter_descriptions
/tb3_1/move_base/DWAPlannerROS/parameter_updates
/tb3_1/move_base/DWAPlannerROS/trajectory_cloud
/tb3_1/move_base/NavfnROS/plan
/tb3_1/move_base/cancel
/tb3_1/move_base/current_goal
/tb3_1/move_base/feedback
/tb3_1/move_base/global_costmap/costmap
/tb3_1/move_base/global_costmap/costmap_updates
/tb3_1/move_base/global_costmap/footprint 
/tb3_1/move_base/global_costmap/inflation_layer/parameter_descriptions
/tb3_1/move_base/global_costmap/inflation_layer/parameter_updates
/tb3_1/move_base/global_costmap/obstacle_layer/parameter_descriptions
/tb3_1/move_base/global_costmap/obstacle_layer/parameter_updates
/tb3_1/move_base/global_costmap/parameter_descriptions
/tb3_1/move_base/global_costmap/parameter_updates  
/tb3_1/move_base/global_costmap/static_layer/parameter_descriptions
/tb3_1/move_base/global_costmap/static_layer/parameter_updates
/tb3_1/move_base/goal
/tb3_1/move_base/local_costmap/costmap
/tb3_1/move_base/local_costmap/costmap_updates
/tb3_1/move_base/local_costmap/footprint
/tb3_1/move_base/local_costmap/inflation_layer/parameter_descriptions
/tb3_1/move_base/local_costmap/inflation_layer/parameter_updates
/tb3_1/move_base/local_costmap/obstacle_layer/parameter_descriptions
/tb3_1/move_base/local_costmap/obstacle_layer/parameter_updates
/tb3_1/move_base/local_costmap/parameter_descriptions
/tb3_1/move_base/local_costmap/parameter_updates
/tb3_1/move_base/parameter_descriptions
/tb3_1/move_base/parameter_updates
/tb3_1/move_base/result
/tb3_1/move_base/status
/tb3_1/move_base_simple/goal
/tb3_1/odom
/tb3_1/scan
/tb3_1/turtlebot3_slam_gmapping/entropy
/tf 
/tf_static

ROBOTIS-Will, 2020-03-23:
@mariocj 
As I know of, the costmap requires a map to initialize.
During the SLAM, there isn't any map file to initialize the costmap, so it will be empty.

mariocj, 2020-03-23:
@ROBOTIS-Will 

Ok, but how can you explain that without using namespaces, using just one turtlebot (let us use tutorial 11.2.1. with turtlebot3_world as example), everything works fine, including local costmal and exploration, and that is by using SLAM method, so there is no loading of any map file. Actually the goal is to create a map, so i must use SLAM method during exploration with turtlebot.

ROBOTIS-Will, 2020-03-25:
@mariocj 
Hi, I'm not sure what the move_base package does, but TurtleBot3 example in 11.2.1 gets user input to control the robot platform for SLAM.
```
roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch
```
When I launch the gazebo and rviz for SLAM explained in eManual, I don't see the local costmap in my rostopic list.

mariocj, 2020-03-25:
@ROBOTIS-Will 

So I tried to set different params for each robot in separate files and now it finally works! Still, I am not sure what was wrong with my original approach, in which i just changed different params through move_base launch file. But maybe i will find that out later. Anyway, thank you very much.

closed 2020-03-26 by ROBOTIS-Will

JDaniel29, 2020-07-24:
So I am having a similar issue where I am trying to run a navigation stack for each robot, but I cannot get the local costmap to update. I tried @mariocj solution where I set different parameters for each robot in separate files and my costmap is still blank. Is there something I am missing? 

All the nodes appear to be connected correctly when I look in the RQT Graph, but when I visualize with RViz the local costmap is a blank white square that is not indicating the wall right next to it. I know the scan topic is working correctly because I am able to run the SLAM nodes and map the house if I manually drive the robot. I also think the rosparam is being set correctly because `rosparam get` results in the correct scan topic being displayed. However, the local costmap itself is still blank. Is there something I am missing here?

Here is what I am running:

Main Launch File (`multi-agent-mapping.launch`)
```
 <!-- Launch the SLAM node, move_base_node and AMCL node for robot_1 -->
  <include ns="$(arg first_tb3)" file="$(find turtlebot3_slam)/launch/turtlebot3_gmapping.launch">
    <arg name="set_base_frame" value="$(arg first_tb3)/base_footprint"/>
    <arg name="set_odom_frame" value="$(arg first_tb3)/odom"/>
    <arg name="set_map_frame" value="$(arg first_tb3)/map"/>
  </include>

  <include ns="$(arg first_tb3)" file="$(find multi-agent-mapping)/launch/move_base.launch">
    <arg name="model" value="$(arg model)" />
    <arg name="move_forward_only" value="$(arg move_forward_only)"/>
    <arg name="cmd_vel_topic" value="cmd_vel"/>
    <arg name="namespace" value="$(arg first_tb3)"/>
  </include>

  <include ns="$(arg first_tb3)" file="$(find multi-agent-mapping)/launch/amcl.launch">
    <arg name="scan_topic" value="$(arg first_tb3)/scan"/>
    <arg name="global_frame_id" value="$(arg first_tb3)/map"/>
    <arg name="base_frame_id" value="$(arg first_tb3)/base_footprint"/>
    <arg name="odom_frame_id" value="$(arg first_tb3)/odom"/>
  </include>
```
`move_base.launch`
```
<launch>
  <!-- Arguments -->
  <arg name="model" default="$(env TURTLEBOT3_MODEL)" doc="model type [burger, waffle, waffle_pi]"/>
  <arg name="cmd_vel_topic" default="/cmd_vel" />
  <arg name="odom_topic" default="odom" />
  <arg name="move_forward_only" default="false"/>

  <arg name="namespace" default=""/>

  <!-- move_base -->
  <node pkg="move_base" type="move_base" respawn="false" name="move_base_node" output="screen">
    <param name="base_local_planner" value="dwa_local_planner/DWAPlannerROS" />
    <rosparam file="$(find multi-agent-mapping)/params/costmap_common_params_$(arg model)_$(arg namespace).yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find multi-agent-mapping)/params/costmap_common_params_$(arg model)_$(arg namespace).yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find multi-agent-mapping)/params/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find multi-agent-mapping)/params/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find multi-agent-mapping)/params/move_base_params.yaml" command="load" />
    <rosparam file="$(find multi-agent-mapping)/params/dwa_local_planner_params_$(arg model).yaml" command="load" />

    <param name="global_costmap/robot_base_frame" value="$(arg namespace)/base_footprint"/>
    <param name="global_costmap/global_frame" value="$(arg namespace)/map"/>
    <param name="local_costmap/robot_base_frame" value="$(arg namespace)/base_footprint"/>
    <param name="local_costmap/global_frame" value="$(arg namespace)/odom"/>

    <remap from="cmd_vel" to="$(arg cmd_vel_topic)"/>
    <remap from="odom" to="$(arg odom_topic)"/>
    <param name="DWAPlannerROS/min_vel_x" value="0.0" if="$(arg move_forward_only)" />
  </node>
</launch>

```

`costmap_common_params.yaml`
```
obstacle_range: 3.0
raytrace_range: 3.5

footprint: [[-0.105, -0.105], [-0.105, 0.105], [0.041, 0.105], [0.041, -0.105]]
#robot_radius: 0.105

inflation_radius: 1.0
cost_scaling_factor: 3.0

map_type: costmap
observation_sources: scan
scan: {sensor_frame: robot_2/base_scan, data_type: LaserScan, topic: robot_2/scan, marking: true, clearing: true}
```