ROBOTIS-GIT/turtlebot3 #546: ROS2 : failed to set clock & change cpu frquency

zhl017, 2020-03-11:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    1.[turtlebot3] ros2 launch turtlebot3_bringup robot.launch.py
    2.[remotePC] ros2 run turtlebot3_teleop teleop_keyboard

8. Copy and Paste your error message on terminal

    - https://imgur.com/vnnLHt3

9. Please, describe detailedly what difficulty you are in

    - brinup is OK, but when I tried to control TB3 with keyboard or another process (ros2 topic echo .... ), it's show this error, and rpi3B+ system crashing(need to reboot).
    - I tried to modify rpi3B+'s cpu frequncy by myself but not success, how can I solved this problem ?

zijian-xu, 2020-03-16:
I met the same issue. Is any available solution to it?

ROBOTIS-Will, 2020-03-20:
@zhl017 @zijian-xu 
It seems like this is power related issue. If you are connecting USB devices to RPi, please remove them and see if it helps to resolve the issue.
Also, make sure that your power supply(SMPS or battery) is providing enough power.
When the battery is low, it may not provide enough power to OpenCR and connected devices.

zijian-xu, 2020-03-20:
@ROBOTIS-Will @zhl017 
Hi Will, thanks you for your reply. I also met one issue when I try to write the firmware to the OpenCR board. Every time after I write the firmware into OpenCR, there is a message like: **cdc_acm 1-1.5:1.0: failed to set dtr/rts.**

Have you guys seen anything like this before? Any idea to solve this issue? Thanks in advance!

ROBOTIS-Will, 2020-03-23:
@zijian-xu 
The USB port might be occupied with something else.
Have you tried [removing the modemmanager](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#remove-modemmanager)?

zijian-xu, 2020-03-23:
@ROBOTIS-Will 
Hi Will, I tried to run the command on the SBC(Ubuntu 18.04). The system shows the modemmanager is not installed.

I used the OpenCR board to develop ROS1 before. Does this have an impact on it?

ROBOTIS-Will, 2020-03-23:
@zijian-xu 
If you have ROS1 TurtleBot firmware on OpenCR, you should also change it to ROS2 firmware.
Please see the [OpenCR setup for ROS2](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/#opencr-setup).

zijian-xu, 2020-03-23:
Hi @ROBOTIS-Will,

I followed this tutorial to write the firmware and got the above errors. Thanks.

ROBOTIS-Will, 2020-03-23:
@zijian-xu 
You might also want to search for similar issues in Raspberry community.
There seems to be many cases that are similar to yours.

zijian-xu, 2020-03-23:
Okay thanks Will.

robotster, 2020-03-23:
Hi,
I have the exact same issue. Got a brand new turtlebot3 burger kit and followed the setup instructions step-by-step for ROS2, in particular
* Remote PC is stock Ubuntu 18.04 with latest updates
* SBC uses recommended Ubuntu 18.04 server for raspberry

Things that I have tried (without success):
* Different power supply options: SMPS and fully charged battery
* Different clock options: default, setting arm_freq=arm_freq_min=1400, setting arm_freq=arm_freq_min=700, force_turbo=0 or 1
* Checking that modemmanager is not installed (it wasn't)
* Using a different RPi 3+ board with the same SD card
* Using ROS2 dashing and eloquent
* Disconnecting power-consuming devices: HDMI, laser scanner
* DHCP on and off (just out of despair...)

No success so far.

robotster, 2020-03-23:
More info:
* After opencr upload, I get "cdc_acm 1-1.1.2:1.0: failed to set dtr/rts". However, the upload itself seems to work, reporting as expected and ending with "[OK] jump_to_fw"
* During teleop operation, the turtlebot moves for a few seconds, then clock related errors kick in:
** bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
** bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
** cpufreq:: __target_index: Failed to change cpu frequency: -22
** hwmon hwmon1: Failed to get throttled (-110)
** [...]
* If the cpu clock is set to a fixed level in /boot/firmware/usercfg.txt, then the first 3 messages disappear, and the errors start with
** hwmon hwmon1: Failed to get throttled (-110)

zijian-xu, 2020-03-24:
@robotster 
My trial results are almost the same as yours. 

If you disconnect your laser scanner when you try to teleop the robot, the robot could be connected longer before the SBC crashed. 

I also try to set the cpu_freq or gpu_freq in /boot/config.txt and it did not work either. I am still looking for other solution in Google. Let's keep updated the situation.

ROBOTIS-Will, 2020-03-24:
@zijian-xu @robotster 
I'll try reinstalling my setup and see if I can replicate the same issue.
Please allow me a few days for this test.

zijian-xu, 2020-03-24:
@ROBOTIS-Will @robotster 
Hi Will, thanks for your help. 

I have been trying to change the scaling governor from ondemand to performance.This seems alleviate the problem. Below is the command that I run:
```sh
systemctl disable ondemand
reboot
```
After running this command, at least in my side, the SBC could run for like 20 mins after the robot is brought up. Teleop is ok as well. But after 20 mins, error like `hwmon hwmon1: Failed to get throttled(-110)` occurred and the SBC crashed.

ROBOTIS-Will, 2020-03-25:
@zijian-xu 
Thank you for the update.
I'll keep this information when I set up my TurtleBot3.

estshorter, 2020-03-26:
I got the same issue.

robotster, 2020-04-01:
@ROBOTIS-Will Did you have a chance to look into this yet?

ROBOTIS-Will, 2020-04-02:
I've installed new ROS2 Dashing on my PC(Ubuntu 18.04) and TurtleBot3(Ubuntu 18.04 Server for Rpi), but didn't get any error related to the RPi cpu.
[OpenCR ROS2 firmware Setup](http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_setup/#opencr-setup) from RPi also didn't have any error either.
Please try with default CPU setting for RPi 3B+.
Thank you.

estshorter, 2020-04-04:
FWIW, my issue is as follows.

1. Which TurtleBot3 you have?

    - [x] Burger

2. Which ROS is working with TurtleBot3?

    - [x] ROS 2 Dashing Diademata

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3+

4. Which OS you installed in SBC?

    - [x] Ubuntu Server 18.04

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 18.04 LTS (Bionic Beaver)

6. Write down software version and firmware version

Sorry, I don't know how to check these versions.

- Software version: [x.x.x]
- Firmware version: [x.x.x]

7. Write down the commands you used in order

[turtlebot3]

- `ros2 launch turtlebot3_bringup robot.launch.py`

[remotePC]

- Open terminal and run the following command
`ros2 run turtlebot3_teleop teleop_keyboard`

- Open another terminal and run the following command
`ros2 launch turtlebot3_bringup rviz2.launch.py`

- Wait for a while (up to 10 minutes)

8. Copy and Paste your error message on terminal
Message such as "Failed to set clock" appears on TB3. (I'm sorry I don't have the log file.)

9. Please, describe detailedly what difficulty you are in
    - TB3 freezes. No reponse for ping and ssh.
    
    Note that I didn't get error such as "Under-voltage detected".
    Also, "OpenCR ROS2 firmware Setup" worked without any error.

keanewatterson, 2020-04-05:
Hello - I'm attaching a stack trace from the moment of crash. I haven't been able to capture proper log output with the error, so (sorry) screen shot of dmesg output. The trace starts at 380.87.

My set-up is like estshorter's above, new TB3, same software versions, same ros2 launch steps leading to a consistent crash repro. Following most of robotster's troubleshooting steps, and zijian-xu's, the crash symptom also persisted.

[image]

ROBOTIS-Will, 2020-04-06:
Thank you for additional information.
I'll try to find out what went wrong with the RPi suspension.

ROBOTIS-Will, 2020-04-08:
Hi all,
From our internal tests, we've also experienced RPi feeze without a clear reason on Ubuntu 18.04.4 server.
In order to resolve this issue, I'd recommend to use 18.04.3 instead.
[http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz](http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz)
Other than the OS version of the Ubuntu server for RPi, all the instructions in the eManual are still valid.
Thank you.

estshorter, 2020-04-08:
@ROBOTIS-Will 
Do you mean that we must not run `sudo apt update && sudo apt upgrade` on Ubuntu 18.04.3?
I think the command above will update the Ubuntu version.

zijian-xu, 2020-04-08:
Hi Will,

Thanks for the update. After reinstalling Ubuntu to 18.03, everything works smoothly. Thank you so much!!!

Best Regards,
Jeff

ROBOTIS-Will, 2020-04-09:
@estshorter 
The `sudo apt update && sudo apt upgrade` only updates and upgrades the package in your current Ubuntu version, but do not update your Ubuntu.
Once you have burned 18.04.3 on the SD card, you can freely use update and upgrade commands.
Thank you.

robotster, 2020-04-14:
Thanks for the updates!
I was already planning to try out 18.04.3 after the reported errors here appeared so *recently*, but didn't find the time during easter holidays.

By the way, be careful with "apt dist-upgrade" which probably will bring you to the next point release.

ROBOTIS-Will, 2020-04-21:
I believe this issue is resolved with above resolution. Please feel free to reopen the issue whenever needed.

closed 2020-04-21 by ROBOTIS-Will

smorad, 2020-10-28:
Googling the error results in this page. If anyone else is having this issue with Ubuntu 20.04, read https://github.com/ROBOTIS-GIT/turtlebot3/issues/593.

TL;DR is make sure you upgrade your kernel to at least `5.4.0-1019.21`
`sudo apt update && sudo apt upgrade`