ROBOTIS-GIT/turtlebot3 #547: ROS2: Failed to write firmware into

zijian-xu, 2020-03-20:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order
cd ~/opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr
ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste your error message on terminal
cdc_acm 1-1.5:1.0:failed to set dtr/rts

9. Please, describe detailedly what difficulty you are in
SBC is connected to a screen via HDMI, OpenCR via micro USB and a Turtlebot3 Lidar.
I used ssh to connect my SBC, every time I typed the above command to update OpenCR, the screen showed the error message above.

And after I brought up turtlebot3, the SBC just crashed within a minute and need to hard reboot.

Thanks!

ROBOTIS-Will, 2020-03-23:
@zijian-xu 
It looks like the USB of RPi is occupied with something else.
Have you tried removing [modemmanager](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#remove-modemmanager)?

zijian-xu, 2020-03-23:
@ROBOTIS-Will
Hi Will, I tried to run the command on the SBC(Ubuntu 18.04). The system shows the modemmanager is not installed.

I used the OpenCR board to develop ROS1 before. Does this have an impact on it?

ROBOTIS-Will, 2020-03-23:
Since this is a duplication of #546, I'll close this thread.
Please continue on #546

closed 2020-03-23 by ROBOTIS-Will