ROBOTIS-GIT/turtlebot3 #558: Burger loses WiFi connectivity and OS crashes after launching robot.launch.py

billmania, 2020-03-24:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

 If you explain which software, which firmware, and where to find these details, I'll provide them.

7. Write down the commands you used in order

    - ros2 launch turtlebot3_bringup robot.launch.py

8. Copy and Paste your error message on terminal

    - watchdog: BUG: soft lockup - CPU# stuck for 22s!
    - Net: No ethernet found. 0 Storage Device(s) found
    - Failed to start Load Kernel Modules.

ubuntu@burger:/var/log$ systemctl status --full systemd-modules-load.service
● systemd-modules-load.service - Load Kernel Modules
   Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-03-23 14:45:16 PDT; 22h ago
     Docs: man:systemd-modules-load.service(8)
           man:modules-load.d(5)
  Process: 789 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
 Main PID: 789 (code=exited, status=1/FAILURE)

Mar 23 14:45:16 burger systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Mar 23 14:45:16 burger systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Mar 23 14:45:16 burger systemd[1]: Failed to start Load Kernel Modules.

9. Please, describe detailedly what difficulty you are in

    - OS repeatedly crashes shortly after starting the launch file.

ROBOTIS-Will, 2020-03-26:
@billmania 
Hi, we are currently taking a look at your issue and trying to reproduce with our setup.
Thank you.

billmania, 2020-03-26:
@ROBOTIS-Will Any progress on this issue? I still have a brand new Turtlebot3 which is unusable.

ROBOTIS-Will, 2020-03-27:
@billmania 
For the errors regarding the systemd-modules-load.service, the Ubuntu server image for Raspberry Pi is missing ib_iser module which is related to iSCSI.
You can avoid the error by commenting out the ib_iser as below.
```
$ sudo nano /lib/modules-load.d/open-iscsi.conf
```
Then, comment out the ib_iser as below
```
iscsi_tcp
#ib_iser
```
Save and reboot will resolve the error.

ROBOTIS-Will, 2020-04-08:
@billmania 
We've also found that Ubuntu 18.04.4 server for RPi is causing unstable operation.
If you are going to use ROS2 Dashing on RPi, I'd recommend to use 18.04.3 server image from the below link.
[http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz](http://old-releases.ubuntu.com/releases/18.04.3/ubuntu-18.04.3-preinstalled-server-arm64+raspi3.img.xz)
Thank you.

ROBOTIS-Will, 2020-04-28:
This thread will be closed as there isn't recent activity on this issue. Please feel free to reopen whenever needed.

closed 2020-04-28 by ROBOTIS-Will