ROBOTIS-GIT/turtlebot3 #561: Please release for Eloquent

SteveMacenski, 2020-03-30:
There is no existing release for eloquent and there are customers asking for it in navigation2: https://github.com/ros-planning/navigation2/issues/1618

ROBOTIS-Will, 2020-04-02:
@SteveMacenski 
Thank you Steve for forwarding the issue.

KimHyunil-0105, 2020-04-02:
I am also curious about release for ROS2 Eloquent. 
Is there a plan to release SW for ROS2 Eloquent version?
If you have a plan, when is the turtlebot3 software for ROS2 Eloquent version released?

andrestoga, 2020-04-26:
Well, they created a branch for Eloquent: eloquent-devel. You can simply install it from source while they release it as a Debian package.

SteveMacenski, 2020-04-27:
@ROBOTIS-Will following up - did you guys manage a release yet?

ROBOTIS-Will, 2020-04-29:
@andrestoga @SteveMacenski Hi, I've created a branch for the eloquent so that I don't forget to release it. A massive binary package updates is scheduled in May along with our eManual instructions. Sorry for the delay!

ROBOTIS-Will, 2020-06-10:
I'm afraid that I have to announce a bad news for the Eloquent developers. We have decided to skip the Eloquent and focus on the Foxy as Eloquent is marked for EOL on this November. We're expecting to be able to deliver TurtleBot3 Foxy contents partially from August. Thank you.

ROBOTIS-Will, 2020-06-12:
@SteveMacenski 
Hi Steve, after the through internal review on this issue, Pyo specifically insisted to keep support on Navigation2 and our developers agreeded to change the plan to release the official Eloquent then Foxy. We’ll begin with simulation part as there have been some unresolved Ubuntu freeze issue with RPi. Thank you very much for your input and appreciate for your interest in our TurtleBot3.
Have a great weekend!

SteveMacenski, 2020-06-12:
I appreciate that, thank you. It may still be worthwhile to get on a call at some point in the next week or so so that I can get an appreciation for your position on the TB3 repos. I noticed this week that the ros2 branches of this haven't moved in several months and it makes me a bit nervous given how tightly coupled it is with Nav2.

andrestoga, 2020-10-14:
@ROBOTIS-Will are you guys still releasing for Eloquent?

ROBOTIS-Will, 2020-12-09:
@andrestoga 
Sorry for the late reply. Eloquent [had merged](https://github.com/ros/rosdistro/pull/26974) on Oct 15th.

closed 2020-12-09 by ROBOTIS-Will