ROBOTIS-GIT/turtlebot3 #562: Error while launching the cartographer SLAM on TB3 in gazebo Simulation

Dhaour9x, 2020-03-31:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

- [ ] Waffle

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3 B+

4. Which OS you installed in SBC?

    - [ ] Ubuntu MATE 16.04 or later

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)

***
7. Write down the commands you used in order

roslaunch turtlebot3_slam turtlebot3_slam.launch 
```
<launch>
  <!-- Arguments -->
  <arg name="model" default="waffle" doc="waffle"/>
  <arg name="slam_methods" default="cartographer" doc="slam type [gmapping, cartographer, hector, karto, frontier_exploration]"/>
  <arg name="configuration_basename" default="turtlebot3_lds_2d.lua"/>
  <arg name="open_rviz" default="true"/>

  <!-- TurtleBot3 -->
  <include file="$(find turtlebot3_bringup)/launch/turtlebot3_remote.launch">
    <arg name="model" value="$(arg model)" />
  </include>

  <!-- SLAM: Gmapping, Cartographer, Hector, Karto, Frontier_exploration, RTAB-Map -->
  <include file="$(find turtlebot3_slam)/launch/turtlebot3_cartographer.launch">
    <arg name="model" value="$(arg model)"/>
    <arg name="configuration_basename" value="$(arg configuration_basename)"/>
  </include>

  <!-- rviz -->
  <group if="$(arg open_rviz)"> 
    <node pkg="rviz" type="rviz" name="rviz" required="true"
          args="-d $(find turtlebot3_slam)/rviz/turtlebot3_$(arg slam_methods).rviz"/>
  </group>
</launch>
```
***
8. Copy and Paste your error message on terminal
```
F0401 00:31:42.201565 63053 sensor_bridge.cc:126] Check failed: sensor_to_tracking->translation().norm() < 1e-5 The IMU frame must be colocated with the tracking frame. Transforming linear acceleration into the tracking frame will otherwise be imprecise.
[FATAL] [1585693902.201913685, 2442.811000000]: F0401 00:31:42.000000 63053 sensor_bridge.cc:126] Check failed: sensor_to_tracking->translation().norm() < 1e-5 The IMU frame must be colocated with the tracking frame. Transforming linear acceleration into the tracking frame will otherwise be imprecise.
[ WARN] [1585693902.246376849, 2442.859000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 2442.86 timeout was 0.1.
*** Check failure stack trace: ***
    @     0x7f4f4b7f55cd  google::LogMessage::Fail()
    @     0x7f4f4b7f7433  google::LogMessage::SendToLog()
    @     0x7f4f4b7f515b  google::LogMessage::Flush()
    @     0x7f4f4b7f7e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x5d5c32  cartographer_ros::SensorBridge::ToImuData()
    @           0x59fbee  cartographer_ros::Node::HandleImuMessage()
    @           0x5af60e  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5bcd49  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f4f4914dd8d  ros::SubscriptionQueue::call()
    @     0x7f4f490f3838  ros::CallbackQueue::callOneCB()
    @     0x7f4f490f523b  ros::CallbackQueue::callAvailable()
    @     0x7f4f49151e39  ros::SingleThreadedSpinner::spin()
    @     0x7f4f49136e9b  ros::spin()
    @           0x5974b4  cartographer_ros::(anonymous namespace)::Run()
    @           0x5944e4  main
    @     0x7f4f47b43830  __libc_start_main
    @           0x596fd9  _start
    @              (nil)  (unknown)
[cartographer_node-3] process has died [pid 63053, exit code -6, cmd /home/fsd/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/fsd/catkin_ws/install_isolated/share/turtlebot3_slam/config -configuration_basename turtlebot3_lds_2d.lua /imu:=/flat_imu __name:=cartographer_node __log:=/home/fsd/.ros/log/03b567d6-733b-11ea-a253-000c29a0d65b/cartographer_node-3.log].
log file: /home/fsd/.ros/log/03b567d6-733b-11ea-a253-000c29a0d65b/cartographer_node-3*.log
[ WARN] [1585693907.574292843, 2447.878000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
```
9. Please, describe detailedly what difficulty you are in

I have created a new gazebo environment(see attachment) and I tried to test cartographer on turtlebot3 in the simulation (virtual environment) , but I get always the same problem.
I have already tried with Gmapping. Hector-SLAM, Karto-SLAM and it works fine.
[image]

[image]

![envigz](https://user-images.githubusercontent.com

ROBOTIS-Will, 2020-04-02:
@Dhaour9x 
Have you also tried setting the "tracking_frame" to base_footprint in the `turtlebot3_lds_2d.lua`? 
It is known that using imu_link causes an issue due to a bug in libgazebo_ros_imu.

Dhaour9x, 2020-04-02:
@ROBOTIS-Will 
I have already tried setting the "tracking_frame" to base_footprint, base_link and Odom but no one does work, I became always the same error. Moreover, I became always a issue after installation of  cartographer stuff when I run catkin_make.
When I run  Gmapping SLAM for the Mapping, I get always a very small image of the map (see image) how can I set the size to get the same image size of  all SLAM algorithm(gmapping, hector, karto, cartographer)

I have a question about the SLAM_method: frontier_exploration,  What is the concept or the technique behind this method?

[image]

pls help

ROBOTIS-Will, 2020-04-03:
@Dhaour9x 
I haven't used other SLAM methods, but there should be an option to set the map resolution.
For example, in Gmapping, you can adjust `delta` parameter .
[http://wiki.ros.org/gmapping#Parameters](http://wiki.ros.org/gmapping#Parameters)

Dhaour9x, 2020-04-03:
thank you @ROBOTIS-Will 
can you help me to solve the cartographer error. Actually I 'm currently writing my master thesis about comparison of 2D-Lidar based SLAM-Algorithmen on Turtlebot3 based on ROS.
The Algorithmen, which i choosed for the comparison are:
* Gmapping
* Hecto-SLAM
* Karto-SLAm
* Cartographer
* Grid-SLAM
I am done with the three first Algorithmen. my main problem right now is the cartographer.
could you please hepl me  to get it work.
thank you

ROBOTIS-Will, 2020-04-06:
@Dhaour9x 
[This posting](https://hackmd.io/@3OnqnT1fTvGgrRoxGGdA-g/r1fAEBGlX?type=view#) might be helpful to resolve your issue.
The posting is based on the issue #217

Dhaour9x, 2020-04-23:
@ROBOTIS-Will thank you. that solve my issue:
the problem was that the cartographer and cartographer_ros (github) were not in the actual release and I always cloned the cartographer stuff under the parent catkin_ws

closed 2020-04-23 by Dhaour9x

SeanZsya, 2021-01-05:
@Dhaour9x Sorry for disturbing you. 
I'm facing the same issue: running cartographer on turtlebot3 waffle in gazebo, with the error message of _Timed out waiting for transform from base_footprint to map to become available before running costmap.._
Actually, I'm trying to use cartographer to replace gmapping in a frontier exploration method of [rrt_exploration](https://github.com/hasauino/rrt_exploration). Cartographer, cartographer_ros & cartographer_turtlebot were built in a separate workspace on my computer.
Would you like to explain your solution more specifically? Did you mean that those packages can't be cloned under the same workspace? They are not in the actual release, how will this affect the result?
Thank you!