ROBOTIS-GIT/turtlebot3 #563: I think there is some typo  at dwa_local_planner_params_model.yaml

minwoominwoominwoo7, 2020-04-01:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [0] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [0] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [0] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [0] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [0] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - HERE
       hello ~ excuse me

I found a typo in the dwa_local_planner_params_X.yaml

https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/master/turtlebot3_navigation/param/dwa_local_planner_params_burger.yaml
max_vel_trans -> max_trans_vel
min_vel_trans -> min_trans_vel

I tried to check if the parameter was reflected by rqt dynamic reconfigure, but the parameter is not reflected.
I think that parameter is a typo.

8. Copy and Paste your error message on terminal

    - This is not erro.  

9. Please, describe detailedly what difficulty you are in

    - HERE there is no difficult, just typo

ROBOTIS-Will, 2020-04-02:
@minwoominwoominwoo7 
Thank you for pointing it out.
The parameter has been deprecated since Melodic.
Please use kinetic-devel branch for the latest code for now(4186fce).

minwoominwoominwoo7, 2020-04-03:
Has the master branch been changed to melodic?
I'm sorry I didn't know if it changed melodic.
Thanks for your answer.  :) 

I think it would be good to change the e manual to a kinetic setting method.
http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/#install-dependent-ros-1-packages
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
-> git clone -b kinetic-devel https://github.com/ROBOTIS-GIT/turtlebot3.git

closed 2020-04-03 by minwoominwoominwoo7

ROBOTIS-Will, 2020-04-08:
@minwoominwoominwoo7 
Thank you for your feedback :)