ROBOTIS-GIT/turtlebot3 #564: TurtleBot3 Bringup Rviz2 sync issue

fizysky, 2020-04-02:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ x ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ x ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ x ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [2.0.1]
    - Firmware version: [0.0.3]

7. Write down the commands you used in order

    [SBC] $ ros2 launch turtlebot3_bringup robot.launch.py
    [REMOTE PC]] $ ros2 launch turtlebot3_bringup rviz2.launch.py

8. Copy and Paste your error message on terminal

    [rviz2-1] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time
    1585833875.72858 but the latest data is at time 1585833875.68246, when looking up transform
    from frame [base_scan] to frame [odom]

9. Please, describe detailedly what difficulty you are in

    I have followed the e-manual [ROS2] for my burger placed on the page 'emanual.robotis.com'.
    Setup and bringup guide and done every step however I came across with this issue when I tried
    to use Rviz2:
    [rviz2-1] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time
    1585833875.72858 but the latest data is at time 1585833875.68246, when looking up transform
    from frame [base_scan] to frame [odom]
    It works fine with other transform, there is chain from odom to base_link without trouble. I also 
    tried configure ntp server and run both commands on RemotePC (after connecting OpenCR and 
    lidar to PC) but the error still occur.

ROBOTIS-Will, 2020-04-08:
@fizysky 
Hi, this is a known issue of Rviz2.
https://github.com/ros2/rviz/issues/359
Thank you.

ROBOTIS-Will, 2020-05-11:
This issue will be closed as there wan't recent activity. Please feel free to reopen this issue when necessary.

closed 2020-05-11 by ROBOTIS-Will