ROBOTIS-GIT/turtlebot3 #565: Turtlebot 3 Navigation Issue

tehnanmanna, 2020-04-09:
After saving the mapping when I launch navigation in the Rviz I am getting the following errors. Please let me know how to solve it. Check the attachment

[image]

roboticslab-fr, 2020-04-17:
Did you check the Timesync between TurtleBot3 and Remote PC ?
I had the same issue. It has been solved once I did the synchronization to NTP server on both TurtleBot and Remote PC :
http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc

ROBOTIS-Will, 2020-05-25:
This issue will be closed as there wasn't recent activity. Please feel free to reopen this when needed.

closed 2020-05-25 by ROBOTIS-Will