ROBOTIS-GIT/turtlebot3 #566: Raw_input in pointop_key raises exception

KomaGR, 2020-04-10:
I'm using gazebo to simulate a TurtleBot3. So far I've been using the **Burger**. When trying to issue navigation instructions to the robot following [this guide](http://emanual.robotis.com/docs/en/platform/turtlebot3/basic_examples/#point-operation) the node kept quitting with the message "shutdown program.". I snooped around in the source code and found that it was probably raising an exception when waiting for user input.

The issue was solved when I changed the referenced line to use `input()` instead of `raw_input()`. Function `raw_input()` is deprecated in python3 and renamed to `input()`. 
https://github.com/ROBOTIS-GIT/turtlebot3/blob/5744eab1c3962c64eb754203180c95ae6858c043/turtlebot3_example/nodes/turtlebot3_pointop_key#L127

If this project does not target python3, I guess this issue can be closed.
<br>
### Side discussion
*The following is not inherently a part of the issue.*

This also raises (pun intended) a good point about how try-catch blocks can hide away problems. While the program fails silently (with a single uninformative error message) with a `try:...except:` block, removing that block lets the program gracefully blow on your face with a pretty informative error:
```
Traceback (most recent call last):
  File "/home/user/catkin_wc/devel/lib/turtlebot3_example/turtlebot3_pointop_key", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/user/catkin_wc/src/turtlebot3/turtlebot3_example/nodes/turtlebot3_pointop_key", line 165, in <module>
    GotoPoint()
  File "/home/user/catkin_wc/src/turtlebot3/turtlebot3_example/nodes/turtlebot3_pointop_key", line 70, in __init__
    (goal_x, goal_y, goal_z) = self.getkey() # This excepts
  File "/home/user/catkin_wc/src/turtlebot3/turtlebot3_example/nodes/turtlebot3_pointop_key", line 135, in getkey
    x, y, z = raw_input("| x | y | z |\n").split()
NameError: name 'raw_input' is not defined
```

ROBOTIS-Will, 2020-04-21:
@KomaGR 
Thank you for your input. Since current TurtleBot3 pointop_key instruction is written based on ROS 1 Kinetic and to avoid possible confusion for someone who followed ROS 1 instruction, we'll handle this deprecated function in ROS 2 section.
Have a great day!

sunghowoo, 2024-11-01:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues, but as this issue has been open for a long time, we will close it for now. Currently, our team is supporting Noetic and Humble versions. Please try again with the latest version, and if you encounter the issue again, feel free to reopen it.

closed 2024-11-01 by sunghowoo