ROBOTIS-GIT/turtlebot3 #571: Turtlebot3 does not move when published on cmd_vel through rostopic

stevenpandiri, 2020-04-20:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x ] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ x] etc (Robotics Dragonboard 845c)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [x ] etc (Ubuntu 16.04 docker container. Host is linux 4.9)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ x] etc (used a single docker container (ubuntu 16.04))

6. Write down software version and firmware version

    - Software version: [linux 4.9]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - rostopic 

8. Copy and Paste your error message on terminal

    - none

9. Please, describe detailedly what difficulty you are in

I have a issue that when published on cmd_vel through rostopic tutlebot3(burger) does not move. 

My configuration:

1. Created 1 docker container with ubuntu 16.04 and ROS kinetic
2. Running ROS master and turtlebot3_core inside the same (single) container. 
3. turtlebot3_core connects to the master
4. rostopic list - lists all the topics correctly
5. when i publish on cmd_vel, turtlebot does not move (rostopic pub -1 /cmd_vel geometry_msgs/Twist -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]')
6. Response on publish - publishing and latching message for 3.0 seconds

My questions: 
Is it correct to run ROS master, turtlebot3_core on the same machine? and 
Also use rostopic to publish to turtlebot3_core from the same container?

I do understand that the setup mentioned in the manual is different but our requirement is such that we cannot use separate SBCs(computers) for remotePC and turtlebot PC. Do you suggest that 2 containers (ubuntu 16.04) can be created one running ROS master and another turtlebot_core and then rostopic publish can be done from inside the container where ROS master is running?

Thanks in advance for your responses

ROBOTIS-Will, 2020-04-21:
@stevenpandiri 
Please be aware that docker may work differently depending on its configuration.
If you have powerful SBC to run everything on board, running all the code in a single PC won't be a problem. If you are using `rotopic pub -1` it will only publish the cmd_vel once so TurtleBot3 may look like it is not reacting with such short command.
Please keep track on the cmd_vel value and see if it continuously providing enough value to move the platform.
Thank you.

stevenpandiri, 2020-04-22:
Thank you Will. It works in a single container, there was some loose connection with the dynamixels. 
Also for the benefit of developers who have everything set fine but still unable to move their tutlebots (i.e. pushing SW1 and SW2 also dont move the turtlebot), this might help..
https://github.com/ROBOTIS-GIT/turtlebot3/issues/263
setting dynamixels:
http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3

ROBOTIS-Will, 2020-04-28:
@stevenpandiri 
When setting up the DYNAMIXEL, you should connect only one to the OpenCR at a time.
After the set up is completed, make sure that you have uploaded the TurtleBot3 core Arduino sketch to the OpenCR.
In order to provide the power to DYNAMIXEL, you should connect battery or SMPS to OpenCR and [turn on the power switch](http://emanual.robotis.com/docs/en/parts/controller/opencr10/#layoutpin-map) near the power input terminal.
Thank you.

ROBOTIS-Will, 2020-05-25:
This issue will be closed as there wasn't recent activity. Please feel free to reopen this when needed.

closed 2020-05-25 by ROBOTIS-Will