ROBOTIS-GIT/turtlebot3 #573: turtlebot can't connect to devices

emma010165, 2020-04-22:
Hi everyone!

1. Which TurtleBot3 you have?
    - Waffle Pi

2. Which ROS is working with TurtleBot3?
    - ROS  Kinetic 

3. Which SBC(Single Board Computer) is working on TurtleBot3?
   - ODROID-XU4

4. Which OS you installed in SBC?
    - [ ] Ubuntu MATE 16.04 or later

9. Please, describe detailedly what difficulty you are in

    - I encountered a problem when trying to reboot the system: the board and the display are not detected anymore. I tried to power up the robot several times but it didn't help. All the lights are on (OpenCR and ODROID-XU4 lights) but no device is connected.
The last thing I did was installing new Linux kernels. I followed the instructions here: [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites]. After reboot, I had HDMI no signal in the display and the mouse and keyboard don't work either.
I don't know if I did something wrong and I am totally blocked.
Can someone please provide some help? 
Thank you.

ROBOTIS-Will, 2020-04-28:
@emma010165 
I'm sorry but there isn't much information for us regarding ODROID-XU4 on TurtleBot3.
Although I'm not quite sure why ODROID isn't working well, but if you are connecting other sensors which requires some power consumption, the OpenCR may not be able to provide sufficient power due to its limitation.
FYI, the maximum power from 12V + 5V + 3.3V port of OpenCR is 54W and if you are drawing too much power from OpenCR, power supply could be unstable.
You might also want to check if there are any similar questions in [ROS Answers](https://answers.ros.org/questions/)
Thank you.

ROBOTIS-Will, 2020-05-25:
This issue will be closed as there wasn't recent activity. Please feel free to reopen this when needed.

closed 2020-05-25 by ROBOTIS-Will