ROBOTIS-GIT/turtlebot3 #578: Turtlebot ROS2-Navigation2 issue amcl waiting for map

iamshyamganatra, 2020-04-25:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?
Nove(simulation)
    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?
NA
    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?
NA
    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py
    -ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=$HOME/turtlebot3_house.yaml

8. Copy and Paste your error message on terminal

    - [bt_navigator-8] [INFO] [bt_navigator_client_node]: "ComputePathToPose" BtActionNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node]: Waiting for "FollowPath" action server
[bt_navigator-8] [INFO] [bt_navigator_client_node]: "FollowPath" BtActionNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node]: Waiting for "/local_costmap/clear_entirely_local_costmap" service
[bt_navigator-8] [INFO] [bt_navigator_client_node]: "ClearEntireCostmap" BtServiceNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node]: Waiting for "/global_costmap/clear_entirely_global_costmap" service
[bt_navigator-8] [INFO] [bt_navigator_client_node]: "ClearEntireCostmap" BtServiceNode initialized
[bt_navigator-8] [INFO] [bt_navigator_client_node]: Waiting for "Spin" action server
[bt_navigator-8] [INFO] [bt_navigator_client_node]: "Spin" BtActionNode initialized
[bt_navigator-8] [INFO] [bt_navigator]: Activating
[lifecycle_manager-1] [INFO] [lifecycle_manager]: The system is active
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[dwb_controller-5] [INFO] [local_costmap.local_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 2059.807 for reason(0)
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [WARN] [amcl]: Waiting for map....

whereas I am getting the map in Rviz

[image]

9. Please, describe detailedly what difficulty you are in

    - when I try to run navigation2.launch.py everything starts correctly but in the end, it shows warning that "waiting for map" and at the same time I have the map on my viz. what is going wrong?

ROBOTIS-Will, 2020-05-07:
@iamshyamganatra 
Hi, did you try the `2D Pose Estimate` and set up the current location of TurtleBot3?
Your issue could be related to [this](https://github.com/ros-planning/navigation2/issues/1042)

ROBOTIS-Will, 2020-09-08:
This issue is closed as there isn't any recent activity. Please feel free to reopen the issue when necessary.
Thank you.

closed 2020-09-08 by ROBOTIS-Will