ROBOTIS-GIT/turtlebot3 #579: Time sync issues when running gmapping

jamescoady, 2020-04-27:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

Waffle Pi

2. Which ROS is working with TurtleBot3?

ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

Raspberry Pi 3

4. Which OS you installed in SBC?

 Raspbian

5. Which OS you installed in Remote PC?

Ubuntu 16.04 LTS (Xenial Xerus)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

**Remote PC**
`Roscore`

**Turtlebot3**
`roslaunch turtlebot3_bringup turtlebot3_robot.launch`

**Remote PC**
`roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping`

8. Copy and Paste your error message on terminal

  `[ WARN] [1587458925.300460191]: Failed to compute laser pose, aborting initialization (Lookup would require extrapolation into the future. Requested time 1587458925.315650008 but the latest data is at time 1587458925.289327081, when looking up transform from frame [odom] to frame [base_footprint])`

9. Please, describe detailedly what difficulty you are in

 I am a beginner with Linux, ROS etc. I have run this gmapping package numerous times a few months ago with no issues.

I have tried to sync both clocks using NTP & Chrony but this does not resolve the issue.

The warning error occurs numerous times before Rviz launches and when I begin to move the robot using teleop keys, the robot loses its position on the map in Rviz and the layout of the room and the points do not represent the space the robot is in.

I have also tried to increase the timing tolerance to 3.0 seconds but this also did not help.

I have completely removed ROS and its dependencies and downloaded ROS again, but this did not fix the issue or warning.

I think it might have something to do with the clocks. I am using dual boot with Windows and have noticed that when I boot Windows OS after using Linux OS, the time on Windows is always 1 hour behind the correct time. Even when I sync this the issue occurs each time.

I have attached the below screenshot, when Rviz does launch, the map takes long to load of the space the robot is in.
[image]

ROBOTIS-Will, 2020-05-07:
@jamescoady 
I see the warning flag in your terminal. Could you reset the OpenCR and try again to see if this helps to resolve the issue?

ROBOTIS-Will, 2020-09-08:
This issue is closed as there isn't any recent activity. Please feel free to reopen the issue when necessary.
Thank you.

closed 2020-09-08 by ROBOTIS-Will