ROBOTIS-GIT/turtlebot3 #580: Turtlebot3 Waffle Pi crash after some time of the follower script execution

bc2606, 2020-04-29:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - [Remote PC] roscore
    - [RPi 3] roslaunch turtlebot3_bringup turtlebot3_robot.launch
    - [Remote PC] roslaunch turtlebot3_follow_filter turtlebot3_follow_filter.launch
    - [Remote PC] roslaunch turtlebot3_follower turtlebot3_follower.launch

8. Copy and Paste your error message on terminal

```
hwmon hwmon1: Failed to get throttled (-110)
mmc1: Timeout waiting for hardware interrupt.
brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
mmc0: timeout waiting for hardware interrupt.
mmc1: Timeout waiting for hardware interrupt.
brcmfmac: brcmf_sdio_htclk: HT Avail request error: -110
ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
rcu= +2-...0: (96 ticks this GP) idle=56a/1/0x4000000000000002 softirq=14557/14558 fqs=2377
```

9. Please, describe detailedly what difficulty you are in

Hello, the follower script works well but sometimes the attached error message appears, and the robot stopped responding. This appears after 1-2min the script launched or just after the last command.

I saw somewhere that it could be some issue with the SD card, but I would like to be sure if any of you already got this issue.

Thanks in advance, and have a good day!

ROBOTIS-Will, 2020-05-11:
@bc2606 
Looks like you are having a similar problem with this case #546.
Since Ubuntu 18.04.4 Server image reported some issue with RPi, it is recommended to use one of the tested OS in the [eManual](http://emanual.robotis.com/docs/en/platform/turtlebot3/raspberry_pi_3_setup/#raspberry-pi-3-setup) or try Ubuntu 18.04.3 Server image which worked for RPi 4.
Thank you.

bc2606, 2020-05-13:
Hello,

I changed to Ubuntu 18.04.3 Server, and since then, everything works perfectly, huge thanks!

closed 2020-05-13 by bc2606