ROBOTIS-GIT/turtlebot3 #581: turtlebot3_navigation2 has a transitive dependency on dynamixel_sdk

ruffsl, 2020-04-30:
It looks like for the ros2 packages, that the dependency relationship `turtlebot3_navigation2` -> `turtlebot3_bringup` -> `turtlebot3_node` requires the installation turtlebot3 drivers even when only `turtlebot3_simulations` is attempted to be used.

https://github.com/ROBOTIS-GIT/turtlebot3/blob/1ff16b4831aa8b4478b2c2e5a59376673635683c/turtlebot3_bringup/package.xml#L22

Is there a reason that `turtlebot3_bringup` needs `turtlebot3_node` or that `turtlebot3_navigation2` needs `turtlebot3_bringup`? Is there a way we could use `turtlebot3_navigation2` with `turtlebot3_simulations` without needing to install hardware drivers?

Related: https://github.com/ROBOTIS-GIT/turtlebot3/issues/481

ruffsl, 2020-04-30:
It also seems that `turtlebot3_gazebo` depends on `turtlebot3` -> `turtlebot3_node`->`dynamixel_sdk`. I think `turtlebot3_gazebo` may want to explicitly depend only on packages it really needs.

ROBOTIS-Will, 2020-05-11:
@ruffsl 
The package is designed for the users who have the TurtleBot3 platform in their hand as our [eManual](http://emanual.robotis.com/docs/en/platform/turtlebot3/overview/) is written with step by step instructions. In the instruction, simulation comes in the later part so the package had to be prepared in the earlier stage. We understand this dependency requires unnecessary package installation for the simulation only users. We'll review this with ROS2 Foxy support and how we can improve our package structure to meet the vast majority of ROS users interest. Thank you for your voice.

sunghowoo, 2024-11-01:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues, but as this issue has been open for a long time, we will close it for now. Currently, our team is supporting Noetic and Humble versions. Please try again with the latest version, and if you encounter the issue again, feel free to reopen it.

closed 2024-11-01 by sunghowoo