ROBOTIS-GIT/turtlebot3 #583: Turtlebot Stops moving in the middle of running then starts moving again with no explanation

AGmochi, 2020-05-08:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - [PC] ssh turtlebot stuff
    - [Turtlebot] turtlebot3_robot.launch
    - [PC] turtlebot3_remote.launch
    - [PC] a rostopic data saving code
    - [PC] turtlebot3_teleop_key.launch

8. Copy and Paste your error message on terminal

    - HERE

9. Please, describe detailedly what difficulty you are in

    - Hello, after I tell the turtlebot a command velocity based off of teleop key, it moves at that velocity for a while before the turtlebot stops on it's own without me telling it to. Afterwards, it starts up on it's own again and immediately back to the original commanded speed. I cannot figure out why signal stopped being sent to the turtlebot so suddenly and why it starts up again. 
I also cannot figure out what kind of signal is being sent to the motors whether it be PWM or something else and how to record it. 
The sudden stops make no sense to me and I can't figure out the solution to it. Any explanation would be much appreciated as well as any possible solution. Thank you.  

I have attached (if the attachment works) the video of the robot stopping without reason. This occurs even when the robot is moving from one part of the room to another. 

[robotstopVid.zip](https://github.com/ROBOTIS-GIT/turtlebot3/files/4601193/robotstopVid.zip)

ashekim, 2020-05-12:
@AGmochi  
Hi AGmochi. Thank you for your questions.
We'll check this issue and get back to you. 

Thank you.

ashekim, 2020-05-13:
@AGmochi 
Hi AGmochi.

Please try the dynamixel firmware recovery.
Enter the link below. The link describes how to dynamixel firmware recovery.

* If you have USB2DYNAMIXEL
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery

* you don't have USB2DYNAMIXEL
use OpenCR and upload Exampes -> OpenCR -> 10.Etc -> usb_to_dxl 
[image]

Please note that all settings will be initialized after firmware recovery.

After trying if you still have the same problem, please let me know.
Thank you.

AGmochi, 2020-05-13:
@ashekim 

Hello Ashekim, 

I tried the dynamixel firmware recovery using the second method. And then uploaded the turtlebot3 firmware through arduino. The problem still persists. I do not think there were any settings that I know of that needed to be specially changed since I mainly just follow the turtlebot3 e-manual when I was setting the turtlebot up.  

One thing I noted was that I tried uploading the turtlebot3 firmware using the shell script method and the problem does not seem to persist but when I upload the firmware using the arduino ide method (with the original code), the robot does the stopping problem. Herein lies another issue to this problem which is that I need to use the arduino ide method to upload the firmware because of the changes I need to make to the source code. 

Is there any idea what may be causing the issue?
Is there something different with the arduino source code that makes it act differently from when using the shell script method? 
And is the signal that is being sent to the motors from opencr a PWM signal?

Thank you

ROBOTIS-Will, 2020-05-18:
@AGmochi 
For a safety reason when a new DYNAMIXEL control signal is not received, DYNAMIXEL will stop rotating. This timeout is set to 500ms in below code and you might want to check if your control signal is in this period.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L85
I cannot assure if this will resolve your issue since you have modified the source code.
Please check your cmd_vel data and publish frequency as well.

AGmochi, 2020-05-29:
Hi, 

I think I may have solved the issue with the stopping which i think was due to adding a raspi camera node which leads me to the next question. Is a raspi camera be compatible with the turtlebot 3 burger? Or is the burger's code or hardware not adapted for the use of a camera?

I can't say I'm familiar with the usage of protocols to fully understand it but I'm guessing it's velocity data signals that gets sent to the dynamixel which then grabs the appropriate amount of current from the battery based off of what the command velocity is?

In addition, is it possible to view the torque applied to motors? If so, how?

Thank you, 
Pamela She

ashekim, 2020-06-01:
@AGE
Hello @AGE. 

I would appreciate it if you could check the link below about raspberrypi camera.
[http://emanual.robotis.com/docs/en/platform/turtlebot3/appendix_raspi_cam/#raspberry-pi-camera](url)

Also, please refer to the link below for the XL-430 torque information.
* Torque ON/OFF 
[http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#p_monitor](url)

* xl430 Performance Graph
[http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#performance-graph](url) 
[http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#present-load126](url)

Thank you.

closed 2020-06-12 by AGmochi