ROBOTIS-GIT/turtlebot3 #590: Gmapping SLAM does not work on a real Turtlebot3 waffle (no map received)

Dhaour9x, 2020-05-14:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [x ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [x ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    -roslaunch turtlebot3_bringup  turtlebot3_robot.launch
- roslaunch turtlebot3_slam turtlebo3_slam slam_methods:=gmapping

8. Copy and Paste your error message on terminal

    
[image]
[image]

    - Hey Guys,
i hope you can help me. I have a real turtlebot3 waffle with an LSD sensor. I am trying to map my office but its not working.
hier are the lauch files of gmapping and turtlebot3_slam:
[turtlebot3_gmapping.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/4629165/turtlebot3_gmapping.txt)
[turtlebot3_slam.txt](https://github.com/ROBOTIS-GIT/turtlebot3/files/4629166/turtlebot3_slam.txt)

after launching  `rosrun rqt_logger_level rqt_logger_level` to DEBUG for more information. I got the following results:
`[DEBUG] [1589467553.058155754]: Connection::drop(2)`
`[DEBUG] [1589467553.058294978]: TCP socket [12] closed`
`[DEBUG] [1589467553.058364267]: Connection::drop(0)`
`[DEBUG] [1589467553.058385182]: Connection::drop(2)`
`[DEBUG] [1589467553.058397898]: Connection::drop(2)`

.

ROBOTIS-Will, 2020-05-21:
Hi,
It looks like your TF is not connected well.
Have you updated source code to the latest version?

Dhaour9x, 2020-05-21:
I  have cloned the GitHub repository from http://emanual.robotis.com/docs/en/platform/turtlebot3/pc_setup/. I don't know if it is the latest version?
can you pls the link of source code to the latest version?
Thank you

ROBOTIS-Will, 2020-05-22:
Hi,
From the eManual, if you are using waffle platform, you can follow below waffle specific command.
- Update OpenCR firmware from TurtleBot3 RPi
```
$ export OPENCR_PORT=/dev/ttyACM0
$ export OPENCR_MODEL=waffle
$ rm -rf ./opencr_update.tar.bz2
$ wget https://github.com/ROBOTIS-GIT/OpenCR-Binaries/raw/master/turtlebot3/ROS1/latest/opencr_update.tar.bz2 && tar -xvf opencr_update.tar.bz2 && cd ./opencr_update && ./update.sh $OPENCR_PORT $OPENCR_MODEL.opencr && cd ..
```
- Update the source code in your PC
```
$ cd ~/catkin_ws/src/turtlebot3
$ git pull
$ cd ~/catkin_ws && catkin_make
```
Then I launched below commands
From PC, 
```
$ roscore
```
from TurtleBot3
```
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
```
from PC,
```
$ export TURTLEBOT3_MODEL=waffle
$ roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=gmapping
```
[image]
[image]

Dhaour9x, 2020-05-22:
I don't know why I have to update OpenCR again.
I can actually control the TB3 from the remote PC using
roslaunch turtlebot3_bringup turtlebot3_robot.launch, but the only problem is when I start gmapping SLAM it does not work. Should I just clone the latest version of the turtlebot repository?

ROBOTIS-Will, 2020-05-25:
Since your `waffle/odom` and `waffle/base_footprint` in the TF tree are separating map and the robot, and I wasn't sure that we are looking at the same source code, so I asked you to update the source code and the firmware.
Also, your TCP connection is failing, check if your wifi signal is stable and does not lose any data.

Dhaour9x, 2020-05-26:
Thank you @ROBOTIS-Will.
I have update the source code and the firmware, but I got always the same issue :(
`[ WARN] [1590485026.536560434]: MessageFilter [target=odom ]: Dropped 100.00% of messages so far. Please turn the [ros.gmapping.message_notifier] rosconsole logger to DEBUG for more information.`

If I check odom and scan topics with `rostopic echo /odom` and `rostopic echo /scan` I got the following results:
```
rostopic echo /odom
`header: 
  seq: 520
  stamp: 
    secs: 1590484856
    nsecs: 179340097
  frame_id: "waffle/odom"
child_frame_id: "waffle/base_footprint"
pose: 
  pose: 
    position: 
      x: 0.00817533861846
      y: 1.14767890409e-05
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0030932368245
      w: 0.999995231628
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: -0.000111956265755
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
---
`
```

rostopic echo /scan
`header: 
  seq: 1537
  stamp: 
    secs: 1590484043
    nsecs: 808229021
  frame_id: "laser"
angle_min: 0.0
angle_max: 6.26573181152
angle_increment: 0.0174532923847
time_increment: 2.98800005112e-05
scan_time: 0.0
range_min: 0.119999997318
range_max: 3.5
ranges: [0.9120000004768372, 1.0010000467300415, 1.0019999742507935, 1.0019999742507935, 0.9319999814033508, 0.8730000257492065, 0.8799999952316284, 0.8870000243186951, 0.878000020980835, 0.8830000162124634, 0.9800000190734863, 1.093999981880188, 1.1119999885559082, 1.1080000400543213, 1.1039999723434448, 1.1069999933242798, 1.1180000305175781, 1.1150000095367432, 1.1260000467300415, 1.1130000352859497, 1.1390000581741333, 1.1160000562667847, 1.128000020980835, 1.1260000467300415, 1.1299999952316284, 1.1269999742507935, 1.1619999408721924, 1.1549999713897705, 1.024999976158142, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0,`
```

Dhaour9x, 2020-05-26:
Thank you @ROBOTIS-Will. After Update it works

closed 2020-05-26 by Dhaour9x