ROBOTIS-GIT/turtlebot3 #593: ROS2 crashing on 18.04.4

russkel, 2020-05-23:
- [x] Burger
    - [x] ROS 2 Eloquent Elusor
    - [x] Raspberry Pi 3
    - [x] Ubuntu Server 18.04.4

After installing as per the manual, I bringup the turtlebot, and then I `ros2 topic echo /battery_state`. After a few moments the raspi crashes. I connected a USB UART to the rpi and collected the errors that occur in order to fix this crash. The following is the output on the UART at crash time:

```
[  516.473656] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  516.482483] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  516.493381] cpufreq: __target_index: Failed to change cpu frequency: -22
[  517.529359] hwmon hwmon1: Failed to get throttled (-110)
[  518.553356] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  518.562141] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  518.573033] cpufreq: __target_index: Failed to change cpu frequency: -22
[  519.609373] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  519.618160] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  519.629058] cpufreq: __target_index: Failed to change cpu frequency: -22
[  521.689367] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  521.701081] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  521.715013] cpufreq: __target_index: Failed to change cpu frequency: -22
[  522.745383] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  522.757209] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  522.771153] cpufreq: __target_index: Failed to change cpu frequency: -22
[  524.797377] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  524.809187] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  524.823065] cpufreq: __target_index: Failed to change cpu frequency: -22
[  525.529343] mmc1: Timeout waiting for hardware interrupt.
[  525.536532] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[  525.849386] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  525.861193] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  525.875088] cpufreq: __target_index: Failed to change cpu frequency: -22
[  526.553345] mmc0: timeout waiting for hardware interrupt.
[  527.897396] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  527.909179] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  527.923064] cpufreq: __target_index: Failed to change cpu frequency: -22
[  528.953410] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  528.965181] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  528.979055] cpufreq: __target_index: Failed to change cpu frequency: -22
[  531.001563] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  531.013347] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  531.027235] cpufreq: __target_index: Failed to change cpu frequency: -22
[  532.057408] bcm2835-cpufreq:bcm2835_cpufreq_set_clock:76: Failed to set clock: 1400000 (-110)
[  532.069194] bcm2835-cpufreq:bcm2835_cpufreq_driver_target_index:176: Error occurred setting a new frequency (1400000)
[  532.083106] cpufreq: __target_index: Failed to change cpu frequency: -22
[  535.769382] mmc1: Timeout waiting for hardware interrupt.
[  536.685378] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  536.692964] rcu: 	0-...0: (24 ticks this GP) idle=376/1/0x4000000000000002 softirq=19318/19319 fqs=2539 
[  536.705606] rcu: 	1-...0: (20 ticks this GP) idle=b22/0/0x1 softirq=18788/18790 fqs=2539 
[  536.797390] mmc0: timeout waiting for hardware interrupt.
[  546.009419] mmc1: Timeout waiting for hardware interrupt.
[  547.033424] mmc0: timeout waiting for hardware interrupt.
[  556.249452] mmc1: Timeout waiting for hardware interrupt.
[  557.273460] mmc0: timeout waiting for hardware interrupt.
[  566.489484] mmc1: Timeout waiting for hardware interrupt.
[  567.513488] mmc0: timeout waiting for hardware interrupt.
[  576.729514] mmc1: Timeout waiting for hardware interrupt.
[  577.753519] mmc0: timeout waiting for hardware interrupt.
[  586.969542] mmc1: Timeout waiting for hardware interrupt.
[  587.993547] mmc0: timeout waiting for hardware interrupt.
[  597.209571] mmc1: Timeout waiting for hardware interrupt.
[  598.233574] mmc0: timeout waiting for hardware interrupt.
[  599.705571] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  599.712548] rcu: 	0-...0: (24 ticks this GP) idle=376/1/0x4000000000000002 softirq=19318/19319 fqs=10164 
[  599.724067] rcu: 	1-...0: (20 ticks this GP) idle=b22/0/0x1 softirq=18788/18790 fqs=10165 
[  607.449599] mmc1: Timeout waiting for hardware interrupt.
[  608.473602] mmc0: timeout waiting for hardware interrupt.
[  617.689623] mmc1: Timeout waiting for hardware interrupt.
[  618.713628] mmc0: timeout waiting for hardware interrupt.
[  628.953653] mmc0: timeout waiting for hardware interrupt.
[  639.193678] mmc0: timeout waiting for hardware interrupt.
[  649.433702] mmc0: timeout waiting for hardware interrupt.
[  659.673725] mmc0: timeout waiting for hardware interrupt.
[  662.725724] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  662.732405] rcu: 	0-...0: (24 ticks this GP) idle=376/1/0x4000000000000002 softirq=19318/19319 fqs=17766 
[  662.743465] rcu: 	1-...0: (20 ticks this GP) idle=b22/0/0x1 softirq=18788/18790 fqs=17767 
[  669.913749] mmc0: timeout waiting for hardware interrupt.
[  680.153774] mmc0: timeout waiting for hardware interrupt.
[  690.393795] mmc0: timeout waiting for hardware interrupt.
[  700.633817] mmc0: timeout waiting for hardware interrupt.
[  710.873841] mmc0: timeout waiting for hardware interrupt.
[  721.113863] mmc0: timeout waiting for hardware interrupt.
[  725.745865] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[  725.752797] rcu: 	0-...0: (24 ticks this GP) idle=376/1/0x4000000000000002 softirq=19318/19319 fqs=25450 
[  725.764263] rcu: 	1-...0: (20 ticks this GP) idle=b22/0/0x1 softirq=18788/18790 fqs=25450 
[  731.353884] mmc0: timeout waiting for hardware interrupt.
[  741.593906] mmc0: timeout waiting for hardware interrupt.
```
The `rcu` and `mmc0` messages repeat until the unit is turned off.

russkel, 2020-05-24:
I have rose a bug with ubuntu-raspi https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1880388

ROBOTIS-Will, 2020-05-25:
@russkel Thank you for attaching the logs. Several customers have also reported that their Ubuntu on RPi does not work and stuck. Ubuntu server 18.04.3 is known to be a working solution for now, but we definitely need Ubuntu 20.04 in order to support ROS 2 Foxy. I'd like to keep this issue thread opened until the resolution is found.
Thank you.

russkel, 2020-06-06:
I am no closer to getting this fixed. If I disconnect both USB plugs it doesn't seem to crash. I'm going to see if I can get the robot working without the LIDAR.

robotster, 2020-06-07:
Confirmed both for 20.04 (foxy) and 18.04.4 (dashing), exactly the same behaviour as described here:
https://github.com/ROBOTIS-GIT/turtlebot3/issues/546

russkel, 2020-06-07:
Thanks @robotster. That issues looks closed with the resolution "only use a very particular out of date version of ubuntu". At least we know there was a change between 18.04.3 and 18.04.4 that broke it.
I don't really know how to compare package versions between Ubuntus.

russkel, 2020-06-17:
I am working around this issue by dumping ubuntu on the RPi and using the newly released Raspberry Pi OS 64b beta. I then run ROS2 using the ros:foxy docker image, using my computer to cross compile the turtlebot3 packages.

juergh, 2020-06-26:
I'm the Ubuntu raspi kernel maintainer and I just started looking into this issue. 18.04.3 uses the 4.15 kernel whereas 18.04.4 uses the 5.3 kernel so this is likely to be a kernel issue. I don't have the HW so this makes debugging challenging. Is there any chance I could get a loaner to track this down?

russkel, 2020-06-28:
Hi @juergh thanks for looking into this. I'm in the Australian region, so unless you're near there I don't think I can ship anything to you that cheaply. I am hoping the ROBOTIS team can help you out, maybe put you in contact with one of their distributors or something @ROBOTIS-Will ?

juergh, 2020-06-28:
Hi @russkel. I learned late last week that our robotics team has a TB3 and managed to reproduce the issue so we should be able to debug this.

russkel, 2020-06-29:
@juergh very pleased to hear this :).

ROBOTIS-Will, 2020-06-30:
@juergh @russkel Thank you for looking into this issue.
Please let me know if there's any help needed during the debugging.
I'm also about to test on my Rpi with Ubuntu server 20.04 and foxy.

juergh, 2020-07-01:
@ROBOTIS-Will Well, our tb3 owner is out this week so if you can reproduce and pull kernel logs that contain the errors that might be a good start. Since it seems to go down hard you probably need to enable kernel logging via the serial console and capture it there. And preferably also some data about what the machine was doing when it fell over.

ROBOTIS-Will, 2020-07-03:
@juergh Got it. Once we collect logs from the frozen machine, we will share it in this thread.

benjaminnicolas, 2020-08-11:
Hi, I'm having quite the same issue with a Raspberry Pi 3 B+ with Ubuntu Server 20.04.
I'm trying to build a small robot with a lidar and a motor controler connected on the USB  ports with ROS1, it works fine with my computer but when I launch the nodes on the RPi it gives me the same hwmon1 error and I can't do anything except unplugging the RPi.

Do you have a fix for this issue ?

russkel, 2020-08-11:
> Hi, I'm having quite the same issue with a Raspberry Pi 3 B+ with Ubuntu Server 20.04.
> I'm trying to build a small robot with a lidar and a motor controler connected on the USB ports with ROS1, it [...]

Check out the Ubuntu bugs link I posted. You can try use the fixed kernel there.

juergh, 2020-08-11:
The fix is committed and works its way through the build/test/release pipeline. We should have an RC kernel later this week.

benjaminnicolas, 2020-08-11:
Ok nice! So I just have to wait until it's released and then do a sudo apt upgrade ?

juergh, 2020-08-11:
> Ok nice! So I just have to wait until it's released and then do a sudo apt upgrade ?

Yes. Or if you want to help with testing (and confirm that the issue is fixed) you can enable the proposed repo and pull it in before it gets released. We typically take 2 weeks for testing before release. I'll comment here once the kernel is in proposed.

russkel, 2020-08-12:
@juergh I just tried the kernel in your PPA, works great. No more crashing. Thanks a lot for sorting that out!

benjaminnicolas, 2020-08-12:
> you can enable the proposed repo and pull it in before it gets released

How can do it ? Sorry I'm kind of new in this

juergh, 2020-08-12:
> > you can enable the proposed repo and pull it in before it gets released
> 
> How can do it ? Sorry I'm kind of new in this

No problem. Add the following line to /etc/apt/sources.list and then run 'apt update && apt upgrade'.
```deb http://ports.ubuntu.com/ubuntu-ports focal-proposed main restricted```

But the new kernel is not in proposed yet.

benjaminnicolas, 2020-08-13:
> But the new kernel is not in proposed yet.

Hi, do you have an approximate idea of when it will be available ? I can't wait to have my robot roaming around :blush:

ThotAlion, 2020-08-13:
I have the same issue in the same conditions (Raspberry Pi 3, Ubuntu-Server 20.04.1, ROS Foxy, Dynamixels). I am going to test the proposed repo.

benjaminnicolas, 2020-08-13:
> @juergh I just tried the kernel in your PPA, works great. No more crashing. Thanks a lot for sorting that out!

I managed to do the same and it works great, thanks !

ThotAlion, 2020-08-13:
@juergh it works also for my configuration, thank you 😊

juergh, 2020-08-18:
Sorry but this change will not land in the current SRU cycle. Setting PREEMPT=y breaks the build of the ZFS kernel module. I need to patch the zfs-linux package first and then SRU it back to Focal before I can make the PREEMPT=y kernel change. Hopefully in the next SRU cycle (in 2 weeks).

ROBOTIS-Will, 2020-08-27:
@juergh Thank you for the updates! We'll test and update our manual once it is released :)

glabmoris, 2020-09-13:
@russkel Do you have a link to the update / kernel config for those of us who are facing the same bug and who'd need to backport it to 18.04 ? 

Thank you!

russkel, 2020-09-14:
@glabmoris you'll have to dig through the commits to the linux-raspi repo. I don't have a link on hand any more, sorry.

glabmoris, 2020-09-14:
Ouch...there goes my sleep hours ;)
Thanks for the reply!

PS: I can also confirm this affects 18.04.5 btw

juergh, 2020-09-14:
@glabmoris This is the commit: https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux-raspi/+git/focal/commit/?h=master-next&id=033763cff70041b3f686733747cfcc6e9d279267

Unfortunately we had to pull that commit from the previous cycle because of an emergency CVE release. The package for the current cycle (which includes the commit again) is currently building (version 5.4.0-1019.21):
https://launchpad.net/~canonical-kernel-team/+archive/ubuntu/ppa/+sourcepub/11578855/+listing-archive-extra

If you're brave, you can install packages from the builder PPA:
$ sudo add-apt-repository ppa:canonical-kernel-team/ppa

Note that the package undergoes a review before it's copied to -proposed and once testing finishes it will migrate to -updates.

Also, the 5.4 raspi kernel in Bionic is a straight backport of the Focal kernel so will naturally inherit that commit but that package has not been built yet as we just started the process of building all kernel packages for the current cycle.

ROBOTIS-Will, 2020-09-14:
In the meantime, we're also testing whether Ubuntu Mate 18.04 and 20.04 works with TurtleBot3.
[https://ubuntu-mate.org/download/](https://ubuntu-mate.org/download/)

juergh, 2020-09-23:
Both Bionic and Focal kernels with this fix are now released. Versions 5.4.0-1019.21{,~18.04.1}. Focal daily images already contain that kernel: http://cdimage.ubuntu.com/ubuntu-server/focal/daily-preinstalled/current/
Bionic images not yet, I need to figure out whom to poke for a rebuild.

russkel, 2020-09-23:
It's a good day. Thank you @juergh.

ROBOTIS-Will, 2020-09-24:
You are the best @juergh.
We'll try them out.

glabmoris, 2020-09-24:
Thank you @juergh ! Is there anything we can do to help make the bionic image rebuild happen?

kyrofa, 2020-09-24:
@glabmoris we'll get it, don't worry. To be clear though, you'll get this if you run `apt upgrade`. As soon as the image is updated, you won't even need to do that, but not having it in the image shouldn't block you at all.

glabmoris, 2020-09-24:
Excellent! Thank you!

russkel, 2020-11-13:
I think this can be closed now.

closed 2020-11-13 by russkel