ROBOTIS-GIT/turtlebot3 #594: Rviz crash on Ros2 Dashing

ChristofGroschke, 2020-05-29:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [x] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [x] etc (Raspberry Pi 4, 4 GB)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [x] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [x] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

   On my Pi:
      export ROS_DOMAIN_ID=30 #TURTLEBOT3
      export TURTLEBOT3_MODEL="burger"
      ros2 launch turtlebot3_bringup robot.launch.py

   On my remote system:
      export ROS_DOMAIN_ID=30 #TURTLEBOT3
      export TURTLEBOT3_MODEL="burger"
      ros2 launch turtlebot3_bringup rviz2.launch.py

8. Copy and Paste your error message on terminal
 
[rviz2-1] rviz2: /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.6/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/include/OgreAxisAlignedBox.h:251: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Promise »(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"« not fulfilled.
[ERROR] [rviz2-1]: process has died [pid 7686, exit code -6, cmd '/opt/ros/dashing/lib/rviz2/rviz2 -d /home/ksc/turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/rviz/model.rviz __node:=rviz2'].

9. Please, describe detailedly what difficulty you are in

I will try to make it easy. This is the standard configuration described in the e-manual. I'm trying to use turtlebot3 with a kinect system + cartographer or rtab-map. This is working great under ros melodic. So i thought upgrading to dashing would be a great idea. The turtlebot in my current configuration is **not** equipped with the kinect i wanted to test the standard configuration first. Only the lidar is installed. The only difference is that ros melodic is installed in parallel on both systems because the kinect driver is not available on ros2 dashing. I want to use the ros1_bridge later on, to pass the relevant pointclouds to ros2. The problem now is, i can't figure out why rviz is crashing. This is not happening when the robot_state_publisher on the Pi is not running ... so my first idea would be a problem with the tf tree published ?

Maybe this is a bug, maybe I'm just missing something.

ROBOTIS-Will, 2020-06-03:
@ChristofGroschke 
Hi, I'm not quite sure what went wrong with your configuration, but below may help to resolve your issue.
It seems like the error is related to the Gazebo.
https://answers.gazebosim.org//question/21441/the-minimum-corner-of-the-box-must-be-less-than-or-equal-to-maximum-corner/
Thank you.

ChristofGroschke, 2020-06-07:
For anyone having similar problems. 
I'm slowly figuring out what is wrong here. I reinstalled everything on a Raspberry Pi3 and it's working fine. So i checked what could be different. On the new Pi only Dashing is installed. On my Desktop: Dashing and Melodic are installed. The urdf looks good. No errors occur with the new Pi and my Desktop machine, even if i include my camera. 

On the Pi4 there seems something terrible wrong the with the two ros versions. When i try to locate the urdf parse library it shows (locate locate liburdf.so):

/opt/ros/melodic/lib/liburdf.so

On my desktop it shows the correct output:
/opt/ros/dashing/lib/liburdf.so
/opt/ros/melodic/lib/liburdf.so

There's something strange going on here. The library itself is installed btw. so it have to be something different.

ChristofGroschke, 2020-06-07:
Last Comment: Solution found. I'm not entirely sure what caused the issue. But it might be useful for you or it is useful to include it in the [tutorials](https://emanual.robotis.com/docs/en/platform/turtlebot3/) FAQ for users with two ROS versions installed ?

I used updatedb on ubuntu to fix the issue above. This [job](http://manpages.ubuntu.com/manpages/xenial/en/man8/updatedb.8.html) is running only once a day normally. It seems to be necessary to do this if you have two ROS versions installed in parallel within this time window before using Colcon to build the turtlebot3 packages ? Otherwise Colcon is linking the wrong liburdf.so within this project ? I didn't checked the build process, it's my first guess ...

Thanks :)

ROBOTIS-Will, 2020-06-10:
@ChristofGroschke Glad that you found the solution. I'm also running melodic and dashing on my Ubuntu18.04 without a problem. 
I didn't get the exact same error with yours, but the lower gazebo version sometimes caused issues like not displaying objects in the window.
Make sure you have the latest gazebo 9 version.
With the `gazebo --version` command, I get below result.
```bash
Gazebo multi-robot simulator, version 9.13.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

ROBOTIS-Will, 2020-09-08:
This issue is closed as there isn't any recent activity. Please feel free to reopen the issue when necessary.
Thank you.

closed 2020-09-08 by ROBOTIS-Will