ROBOTIS-GIT/turtlebot3 #595: How can I get the same map size for all SLAM algorithms?

Dhaour9x, 2020-05-30:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?
    - [x ] Waffle

2. Which ROS is working with TurtleBot3?

    - [ x] ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x] Raspberry Pi 3

4. Which OS you installed in SBC?

    - [ x] Ubuntu MATE 16.04 or later

5. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)

I 'm currently trying to compare the maps of SLAM algorithms for Alignment, Therefor The SLAM maps size must be the same to get meaningful results.
   I have tried to create maps of the gazebo Environment with the following SLAM Algorithms (gmapping, hector, karto and cartographer) on TB3 waffle.
I got very different map size of the generated SLAM maps. Although resolution value is 0.05 for all SLAM algorithms.
here the size of SLAM maps:

SLAM |Gmapping SLAM | Hector SLAM | karto SLAM | Cartographer
----------- | ------------ | ------------- | ------------- | ---------------
Width | 384| 2048 | 71 | 61
Height | 384| 2048 | 82 | 99
Maps | [image] | [image] | [image] |  [image]

how can I get  the same Maps size of all SLAM algorithms. for example like this 
[image]

I have used a Image editor such as **ImageMagik** to resize the maps, but it has bad impact on the shape of the Environment. 

I have used the following commands for each SALM.
    - roslaunch turtlebot3_gazebo navibot.launch
[image]#
    - roslaunch turtlebot3_slam turtlebot3_slam slam_methods:=gmapping
    - roslaunch turtlebot3_slam turtlebot3_slam slam_methods:=hector
    - roslaunch turtlebot3_slam turtlebot3_slam slam_methods:=karto
    - roslaunch turtlebot3_slam turtlebot3_slam slam_methods:=cartographer

I hope you can help me :)

ROBOTIS-Will, 2020-06-03:
Hi,
The map images are created by their own SLAM algorithm and I don't think we can make them to give us the same result for the identical environment. 
I could be wrong as I'm not an expert on SLAM.
It looks like we can easily manipulate the size of Hector map by adjusting the `map_size` parameter in the `turtlebot3_hector.launch` file. 
However, this will not allow us to have a rectangular map image like the Cartographer as the map size of Hector mapping is already defined [here](http://wiki.ros.org/hector_mapping/#Parameters) with map_size * map_size square grid.
Thank you.

Dhaour9x, 2020-06-13:
In this situation, where the maps that I got are not having the same size, I have to perform "Image registration" https://github.com/Dhaour9x/Image-Registration. The assumption is, I try to align map2, map3 and map4, with the same scale as Ground truth map.

my question is, does this method make sense to solve the problem and is the comparison between ground truth map and registered (edited) maps reliable?? 

Thank you

ROBOTIS-Will, 2020-07-14:
I'm sorry for the late reply.
As long as you can keep the map resolution same as original file, modifying the image should be fine.
Each pixel is defined to represent certain distance in each algorithm.
Thank you.

Dhaour9x, 2020-07-22:
Thank you @ROBOTIS-Will

closed 2020-07-22 by Dhaour9x