ROBOTIS-GIT/turtlebot3 #596: Using frontier exploration SLAM

ozangungortuhh, 2020-05-31:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [X] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [X] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [X] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [X] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [X] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - roslaunch turtlebot3_gazebo turtlebot3_world.launch
    - roslaunch turtlebot3_slam turtlebot3_slam.launch slam_methods:=frontier_exploration

8. Copy and Paste your error message on terminal

    - HERE

9. Please, describe detailedly what difficulty you are in

- I am trying to run frontier exploration SLAM in turtlebot world from gazebo but having some troubles to understand the basic concept of using the package. In http://wiki.ros.org/frontier_exploration, it is written and displayed that a user-defined polygon should be given to the frontier exploration and it starts exploring that defined area. However, when I click on Publish Point on rviz and select a point on the map, nothing actually happens. I can' see the selected point therefore I can't create a polygon for the exploration to start. 

- I am quite new with SLAM and might be lacking some basic theoretical knowledge. Another question is, from my understanding AMCL is localizing the robot within a pre-defined map, so why is there both gmapping and amcl nodes running while executing frontier exploration? What I want to do is exploring a completely unknown area autonomously so I did not understand why we should have AMCL node as well.

Thanks a lot.

ozangungortuhh, 2020-05-31:
I think my problem is that I am using ROS Melodic with Ubuntu 18.04 and frontier exploration package is only available for ROS Kinetic. If anyone has experience making the frontier_exploration package with Ubuntu 18.04, I would appreciate if he/she could comment :)

ROBOTIS-Will, 2020-06-04:
@ozangungortuhh 
Hi, I've created an issue thread for Melodic support. You might want to try the `melodic-devel` branch that contains some changes to support Melodic.
https://github.com/paulbovbel/frontier_exploration/commit/a6a3854504719754beccdb2f4879592a894b59b0

closed 2020-06-04 by ROBOTIS-Will