ROBOTIS-GIT/turtlebot3 #599: Foxy Release

spragunr, 2020-06-09:
I'm just wondering when there will be an official Foxy release?  I see that there is a foxy_devel branch.  I just received several TB3 that I will be using for a course in the fall.  Working with our sysadmin will be simplified if there are deb packages available. I see #561 is open, but Eloquent is EOL in just a few months.

ROBOTIS-Will, 2020-06-10:
Hi,
We're expecting that the Foxy support will be available from August.
As you already know, Eloquent will not be supported while we are focusing on Foxy.

rafaelhuang-intel, 2020-09-21:
Hi ,

Doed any update of foxy release ? 

Currently, Yocto only found dashing version (.0.2.6) under Yocto Dunfell, and I also hope to know what time we can use foxy version on Yocto?

ruffsl, 2020-09-22:
Looks like the turtlebot3 packages are still missing from the foxy release, making it tricky to support ROS2 Foxy for the upcoming fall semester:

http://repo.ros2.org/status_page/compare_dashing_foxy.html

Indeed, it would be helpful to have an educational robot platform released into the current ROS2 LTS.

ROBOTIS-Will, 2020-09-23:
Hi @rafaelhuang-intel @ruffsl ,
Thank you all for the interest in this issue.
We're currently working/testing on Foxy with Raspberry Pi3B+, but having some issues with Ubuntu crashing(#593).
I'll try to release the Foxy as soon as possible.
We'll review to release on Yocto.

andrestoga, 2020-10-14:
@ROBOTIS-Will Did you guys already test it with the kernel fixed in Focal?

SebO-L, 2020-10-20:
Hi, 
I'm a ROS2 newcomer and I was waiting for the foxy release. I'd like to report a successful installation (burger bot)
- PC and raspberry on focal &ros2 foxy 
- Turtlebot packages install with the foxy-devel branch.

So far the navigation is up. I have been able to save the map and navigate (odometry ok)
Thanks for our work

DLu, 2020-10-20:
It might be worthwhile to do a release of the parts that are working (i.e. the description) so things like the navigation tutorials can be used in Foxy, even if the real robot cannot.

DLu, 2020-10-20:
This is probably closed via https://github.com/ros/rosdistro/pull/26975, right?

ROBOTIS-Will, 2020-11-12:
@DLu Yes, TurtleBot3 Foxy package has been released.
We've tested it with the latest Ubuntu 20.04 preinstalled server image for Raspberry Pi 3B+ and preparing to support RPi 4B in the next year.

rafaelhuang-intel, 2020-11-18:
I meet a issue of turtlebot3 navigaiton & Rviz (https://github.com/ROBOTIS-GIT/turtlebot3/issues/677 ), does anyone meet same?

ROBOTIS-Will, 2020-12-01:
This issue thread will be closed as TurtleBot3 Foxy has been released. Please feel free to reopen if needed.

closed 2020-12-01 by ROBOTIS-Will