ROBOTIS-GIT/turtlebot3 #601: turtlebot3_node does not specify dependency on dynamixel_sdk

kdhansen, 2020-06-17:
I am setting up a simulation environment for TB3 with Eloquent on Focal. I have checked out the eloquent-devel branch. I ran `rosdep` to resolve dependencies, however dynamixel_sdk was not specified.

`colcon build` returned:
```
--- stderr: turtlebot3_node                                                                            
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Finddynamixel_sdk.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "dynamixel_sdk", but CMake did not find one.

  Could not find a package configuration file provided by "dynamixel_sdk"
  with any of the following names:

    dynamixel_sdkConfig.cmake
    dynamixel_sdk-config.cmake

  Add the installation prefix of "dynamixel_sdk" to CMAKE_PREFIX_PATH or set
  "dynamixel_sdk_DIR" to a directory containing one of the above files.  If
  "dynamixel_sdk" provides a separate development package or SDK, be sure it
  has been installed.

---
```

ROBOTIS-Will, 2020-06-30:
@kdhansen 
Thank you for reporting the issue. I'll be preparing to release eloquent dynamixel_sdk. 
This may take a couple weeks to be released. Could you try building the `eloquent-devel` source branch of the DynamixelSDK until then?
```
$ mkdir -p ~/turtlebot3_ws/src && cd ~/turtlebot3_ws/src
$ git clone -b eloquent-devel https://github.com/ROBOTIS-GIT/DynamixelSDK.git
```
Thank you.

ROBOTIS-Will, 2020-09-08:
This issue thread will be closed as DynamixelSDK binary was released for Eloquent.

closed 2020-09-08 by ROBOTIS-Will