ROBOTIS-GIT/turtlebot3 #602: Turtlebot3 jerking while moving along navigation path.

Necromantia000, 2020-06-18:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?
    - [ ] ROS 2 Dashing Diademata

3. Which SBC(Single Board Computer) is working on TurtleBot3?
    - [ ] Raspberry Pi 3

4. Which OS you installed in SBC?
    - [ ] Ubuntu Server 18.04 or later

5. Which OS you installed in Remote PC?
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

   ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=$HOME/map.yaml

8. Copy and Paste your error message on terminal

    [rviz2-9] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  (repeat 1000 times)

9. Please, describe detailedly what difficulty you are in

    - While in RViz, after selecting navigation goal, Turtlebot3 jerks constantly while attempting to move along the path.  Upon zooming into the path selected, there are numerous arrows showing where it should jerk.
I've tried going through the waffle_pi.yaml looking for anything out of the ordinary and ran comparisons to others to no avail.  Teleop works fine.
[VID_20200618_153113.zip](https://github.com/ROBOTIS-GIT/turtlebot3/files/4801038/VID_20200618_153113.zip)

waffle_pi.yaml
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05

amcl_map_client:
  ros__parameters:
    use_sim_time: False

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False

bt_navigator:
  ros__parameters:
    use_sim_time: False
    bt_xml_filename: "bt_navigator.xml"

dwb_controller:
  ros__parameters:
    use_sim_time: False
    debug_trajectory_details: True
    min_vel_x: 0.0
    min_vel_y: 0.0
    max_vel_x: 0.26
    max_vel_y: 0.0
    max_vel_theta: 1.0
    min_speed_xy: 0.0
    max_speed_xy: 0.26
    min_speed_theta: 0.0
    min_x_velocity_threshold: 0.001
    # Add high threshold velocity for turtlebot 3 issue.
    # https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/75
    min_y_velocity_threshold: 0.5
    min_theta_velocity_threshold: 0.001
    acc_lim_x: 2.5
    acc_lim_y: 0.0
    acc_lim_theta: 3.2
    decel_lim_x: -2.5
    decel_lim_y: 0.0
    decel_lim_theta: -3.2
    vx_samples: 20
    vy_samples: 5
    vtheta_samples: 20
    sim_time: 1.7
    linear_granularity: 0.05
    xy_goal_tolerance: 0.25
    transform_tolerance: 0.2
    critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist", "GoalDist"]
    BaseObstacle.scale: 0.02
    PathAlign.scale: 0.0
    GoalAlign.scale: 0.0
    PathDist.scale: 32.0
    GoalDist.scale: 24.0
    RotateToGoal.scale: 32.0

local_costmap:
  local_costmap:
    ros__parameters:
      use_sim_time: False
      global_frame: odom
      plugin_names: ["obstacle_layer", "inflation_layer"]
      plugin_types: ["nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::InflationLayer"]
      rolling_window: true
      width: 3
      height: 3
      resolution: 0.05
      robot_radius: 0.22
      inflation_layer.cost_scaling_factor: 20.0
      obstacle_layer:
        enabled: True
      always_send_full_costmap: True
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
  local_costmap_client:
    ros__parameters:
      use_sim_time: False
  local_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: False

global_costmap:
  global_costmap:
    ros__parameters:
      use_sim_time: False
      robot_radius: 0.22
      obstacle_layer:
        enabled: True
      always_send_full_costmap: True
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
  global_costmap_client:
    ros__parameters:
      use_sim_time: False
  global_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: False

map_server:
  ros__parameters:
    use_sim_time: False
    yaml_filename: "map.yaml"

lifecycle_manager:
  ros__parameters:
    use_sim_time: False
    autostart: True
    node_names: ['map_server', 'amcl',
                 'world_model', 'dwb_controller',
                 'navfn_planner', 'bt_navigator']

lifecycle_manager_service_client:
  ros__parameters:
    use_sim_time: False

lifecycle_manager_client_service_client:
  ros__parameters:
    use_sim_time: False

navfn_planner:
  ros__parameters:
    use_sim_time: False
    tolerance: 0.0
    use_astar: false

navfn_planner_GetCostmap_client:
  ros__parameters:
    use_sim_time: False

robot_state_publisher:
  ros__parameters:
    use_sim_time: False

world_model:
  ros__parameters:
    use_sim_time: False
[image]

cs@robotis.com thanks for the help

Necromantia000, 2020-06-23:
After calling Robotis and being told to leave my issue here in the Github forum, I now know that there isn't any help a new user with this platform and I feel sorry for any person that spent money on this system.

ROBOTIS-Will, 2020-06-30:
@Necromantia000 I'm sorry about the delayed reply on this.
When you launched the navigation, did you perform the `2D Pose Estimate`?
Since small green arrows are wide spread, make sure to do the [initial pose estimation](https://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_navigation2/#estimate-initial-pose) so that the robot can determine its present pose.
Thank you.

Necromantia000, 2020-06-30:
Pose was inputted.   Seems to be happening only when in navigation mode.

MikeHallettUK, 2020-11-08:
I agree that I find move-base to be "jerky" with my Turtlebot3 burger.

To start with, the clockwise lidar is not corrected properly - see about fix @ [How to tf scans to PointCloud2 while TurtleBot3 LDS Lidar rotates clockwise ? ](https://github.com/ROBOTIS-GIT/turtlebot3/issues/633)

and check if your [Turtlebot3 IMU and ODOM yaw is non-zero on Startup](https://github.com/ROBOTIS-GIT/turtlebot3/issues/655)

.... either of which will cause trouble "navigating".

ROBOTIS-Will, 2021-01-29:
There was a jerky TurtleBot3 simulation issue with ROS2 Foxy, and the [fix](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/commit/dcfc1c720e5c618adb2e7f30877f85371815faec) was released.
Not sure if this will fix the same for the Dashing as I couldn't replicate the issue on my Dashing setup.
Please feel free to reopen this when necessary.

closed 2021-01-29 by ROBOTIS-Will