ROBOTIS-GIT/turtlebot3 #607: SLAM algorithms doesn't work on a bagfile

Dhaour9x, 2020-06-23:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ ] Burger
    - [x ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ x] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ x] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [x ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ x] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

7. Write down the commands you used in order

```
$ roscore
$ export TURTLEBOT3_MODEL=waffle
$ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ export TURTLEBOT3_MODEL=waffle
$ rosrun rviz rviz -d `rospack find turtlebot3_slam`/rviz/turtlebot3_slam.rviz
$ roscd turtlebot3_slam/bag
$ rosbag play ./ground.bag
```

8. Copy and Paste your error message on terminal
[image]

 
```rosrun rviz rviz -d `rospack find turtlebot3_slam`/rviz/turtlebot3_slam.rviz```

```
[ INFO] [1592939181.850658835]: rviz version 1.12.17
[ INFO] [1592939181.850723068]: compiled against Qt version 5.5.1
[ INFO] [1592939181.850731910]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1592939181.994656277]: Stereo is NOT SUPPORTED
[ INFO] [1592939181.994763535]: OpenGl version: 3 (GLSL 1.3).
```

9. Please, describe detailedly what difficulty you are in
I have tried to execute different SLAM using .bag file, which I have already recorded [image]
I have recorded bag file using the following commands:

1. `roslaunch turtlebot3_gazebo navibot_world.launch`

[image]

2. `roslaunch turtlebot3_teleop turtlebot3_teleop_key.launch `
3. `rosbag record -o ground /scan /tf  /odom`

I hope you can help me

ROBOTIS-Will, 2020-06-30:
@Dhaour9x Hi, please allow me some time to review this. I'll get back to you as soon as possible. Thanks!

Dhaour9x, 2020-06-30:
Thank you @ROBOTIS-Will 
I am looking forward to your support 👍 -)

ROBOTIS-Will, 2020-07-06:
Hi, I'm sorry about the late reply.
One of our engineers suggested to record the map data, since fixed frame is based on the map.
The tf may not be connected among scan, odom and the map.

Dhaour9x, 2020-07-07:
Thank you for your feedback @ROBOTIS-Will 
I have recorded a bag  file ![groundTh.bag](https://drive.google.com/file/d/1cDEkcxOhhCL0w9krr1ozmeZVCELl7JsN/view?usp=sharing
with following topics: `/tf  /odom /map /scan.`  The First issue is the message timestamps on the bag are set to the beginning of UNIX time (Jan 1 1970). 
`rosbag info groundTh.bag`
```
path:        groundTh.bag
version:     2.0
duration:    6:39s (399s)
start:       Jan 01 1970 01:09:21.55 (561.55)
end:         Jan 01 1970 01:16:00.66 (960.66)
size:        16.4 MB
messages:    25943
compression: none [22/22 chunks]
types:       nav_msgs/Odometry     [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/LaserScan [90c7ef2dc6895d81024acba2ac42f369]
             tf2_msgs/TFMessage    [94810edda583a504dfda3829e70d7eec]
topics:      /odom   11974 msgs    : nav_msgs/Odometry    
             /scan    1995 msgs    : sensor_msgs/LaserScan
             /tf     11974 msgs    : tf2_msgs/TFMessage
```

 How can I correct the timestamps? I'm trying to launch the .bag file in the same SLAM launch file, How can I do that in `turtlebot3_slam.launch` file

Another Issue is the wheels link have no tf (see pic)
[image]

So for launching SLAM on bag file I have used the following commands:
```
roslaunch turtlebot3_slam turtlebot3_slam slam_methods:=gmapping 
rosbag play groundTh.bag --clock
```
Is this correct or are there other launch files I need to run too??

Thank you

Dhaour9x, 2020-07-12:
@ROBOTIS-Will can you please give me the solution for this issue.
I am looking forward to your reply

Dhaour9x, 2020-07-21:
@ROBOTIS-Will 
please I need your help to solve this issue 
Thank you

ROBOTIS-Will, 2020-07-24:
@Dhaour9x I'm sorry about the delayed response. I'll try to replicate your issue and get back to you. Thank you for your patience!

Dhaour9x, 2020-07-24:
Thank you @ROBOTIS-Will 
I'm looking forward for your feedback

ROBOTIS-Will, 2020-12-09:
@Dhaour9x 
I'm sorry for the late check in, but have you resolved this issue?

Dhaour9x, 2020-12-09:
not yet, I still have the issue

sunghowoo, 2024-11-01:
Apologies for the delayed response. We have assembled a TF team to address TurtleBot3 issues, but as this issue has been open for a long time, we will close it for now. Currently, our team is supporting Noetic and Humble versions. Please try again with the latest version, and if you encounter the issue again, feel free to reopen it.

closed 2024-11-01 by sunghowoo