ROBOTIS-GIT/turtlebot3 #610: "No transform from [wheel of left / right]" on rviz.

net0105net, 2020-06-26:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [○] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [○ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [○ ] Raspberry Pi 3 B+
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [○ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [○ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [1.2.3]
    - Firmware version: [1.4.15]

7. Write down the commands you used in order

    roscore
    roslaunch turtlebot3_bringup turtlebot3_robot.launch
    roslaunch turtlebot3_bringup turtlebot3_remote.launch
    roslaunch turtlebot3_example interactive_markers.launch
    rosrun rviz rviz -d `rospack find turtlebot3_example`/rviz/turtlebot3_interactive.rviz

8. Copy and Paste your error message on terminal
   
    (nothing)

9. Please, describe detailedly what difficulty you are in

   I run above commands, but there are error of " No transform from [wheel of left / right] " on rviz and Turtlebot does not move.

   I found similar issue related on IP address, but I could not find any problem.

   I am Japanese, so I'm sorry that my English not so good.
   I attach some screen shot. "rel" is remote PC, and "kakeru" is Turtlebot's PC.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

ROBOTIS-Will, 2020-06-30:
@net0105net 
Hi, thank you for giving us specific details and captured images.
In case of getting Failed to get param error, most likely the OpenCR has problem with connecting to Raspberry Pi.
Please reboot OpenCR board with the reset button and see if it is detected from the Raspberry Pi port `/tty/ACM0`
[image]

net0105net, 2020-06-30:
@ROBOTIS-Will 

Thank you for reply.

I reboot OPENCR and change mode of Raspberry Pi port, so “Failed to get param” error fade, but there are still error on rviz and I can’t control turtlebot.

net0105net, 2020-07-05:
I found additional error related rviz. When I run rviz, the error displayed like below.

[ WARN] [1593935337.447896351]: Interactive marker 'turtlebot3_marker' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

ROBOTIS-Will, 2020-07-27:
@net0105net 
I'm so sorry about the delayed response.
I just realized that you might be running different ROS versions for Remote PC and TurtleBot3 as your Linux version doesn't match.
The recommended OS for Melodic Morenia is Ubuntu 18.04. 
Since we haven't fully tested with 16.04 + Melodic combination, there could be incompatibility.

net0105net, 2020-07-28:
@ROBOTIS-Will 
I found TB3 burger has RaspberryPi 3 B+ now and should be installed UbuntuMATE in official tutorial, so I installed UbutuMATE 18.04 and ROS Melodic.

Should I change the OS of RasberryPi ? 
( Or what should I do? )

ROBOTIS-Will, 2020-08-12:
Hi,
If you have installed 18.04, please install ROS 1 Melodic on your Raspberry Pi as well
Thank you.

ROBOTIS-Will, 2020-09-08:
This issue is closed as there isn't any recent activity. Please feel free to reopen the issue when necessary.
Thank you.

closed 2020-09-08 by ROBOTIS-Will