ROBOTIS-GIT/turtlebot3 #616: The speed of turtlebot3 is too slow

alexis-pepe, 2020-07-13:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [√] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [√] ROS 1 Kinetic Kame
    - [ ] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [ ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [√] etc (PLEASE, WRITE DOWN YOUR SBC HERE)
TX2
4. Which OS you installed in SBC?

    - [ ] Raspbian
    - [√] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [√] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

6. Write down software version and firmware version

    - Software version: [Ubuntu 16.04 LTS]
    - Firmware version: turtlebot3]

7. Write down the commands you used in order

    -  speed

8. Copy and Paste your error message on terminal
[image]

    -  too slow

9. Please, describe detailedly what difficulty you are in

    -  The walking speed of turtlebot3 robot is too slow, only 22m/s . Can the movement speed be improved through the Dynamixel code here? We need a fast experimental scene, More convincing experiments

ROBOTIS-Will, 2020-07-14:
Hi,
[The translation velocity of TurtleBot3](https://emanual.robotis.com/docs/en/platform/turtlebot3/specifications/#hardware-specifications) platform is dependent to the DYNAMIXEL and cannot exceed due to the reduction gear ratio.
Thank you.

ROBOTIS-Will, 2020-09-08:
This issue is closed as there isn't any recent activity. Please feel free to reopen the issue when necessary.
Thank you.

closed 2020-09-08 by ROBOTIS-Will