ROBOTIS-GIT/turtlebot3 #618: Extrapolation error when following the tutorial of navigation2

ClarkyMacLeod, 2020-07-16:
[x] Waffle Pi
    [x] ROS 2 Dashing Diademata
    [x] Raspberry Pi 3
    [x] Ubuntu Server 18.04 or later
    [x] Ubuntu 18.04 LTS (Bionic Beaver)

I followed the Turtlebot3 Tutorial to create a map with SLAM and use it to follow the Tutorial on Navigation2.

    $ ros2 launch turtlebot3_bringup robot.launch.py
    $ ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=$HOME/map.yaml

I set the estimated 2d pose in rviz, then the following error message is posted in the rviz-terminal:

    [rviz2-9] [ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1594895008.39042 but the 
    latest data is at time 1594895008.34734, when looking up transform from frame [base_scan] to frame [map]

Now, when I set a goal, the path is calculated and drawn, but the robot only moves in abrupt moves and the navigation is not completed.

ROBOTIS-Will, 2020-07-21:
@ClarkyMacLeod 
Hi, thank you for reporting the issue. We'll take a look at it and get back to you.
Thank you for your patience.

ashekim, 2020-09-08:
@ClarkyMacLeod 
Hi. Thanks for your interest our TB3!

Please check Timesync between your TurtleBot3 and Remote PC?
Please refer to the [33.2 Timesync between TurtleBot3 and Remote PC](https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc) shown in the eManual.

Test above and let me know if it still doesn't work,
Thank you.

ClarkyMacLeod, 2020-09-21:
> Please check Timesync between your TurtleBot3 and Remote PC?
> Please refer to the [33.2 Timesync between TurtleBot3 and Remote [...]

Hi ashekim,
yes I tried the Timesync with ntp.ubuntu.com, also I tried to use the remote PC as timesync server.
It still doesn't work.

ashekim, 2020-09-22:
@ClarkyMacLeod 

Hi :)

Would you check frequency of tf topics using below commands?
$ ros2 topic hz /tf

OpenCR publishes tf topics in 30Hz. So, If the frequency is significantly different, check your OpenCR and network environments.

I attach the link below. Thank you for your reference.
[ref 1.](https://answers.ros.org/question/188046/tf-lookup-would-require-extrapolation-into-the-future/)
[ref 2.](https://answers.ros.org/question/357836/nav2-teb-lookup-would-require-extrapolation-into-the-future/)

And it would be nice to ask questions on the ROS [forum](https://answers.ros.org/questions/) as well.
Thank you.

ROBOTIS-Will, 2020-12-01:
This issue thread will be closed as there is no recent activity. Please feel free to reopen if needed.

closed 2020-12-01 by ROBOTIS-Will