ROBOTIS-GIT/turtlebot3 #621: Does Turtlebot3 use EKF to improve Odometry ?

MikeHallettUK, 2020-07-17:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?

    - [ x] Burger
    - [ ] Waffle
    - [ ] Waffle Pi

2. Which ROS is working with TurtleBot3?

    - [ ] ROS 1 Kinetic Kame
    - [ x] ROS 1 Melodic Morenia
    - [ ] ROS 2 Dashing Diademata
    - [ ] ROS 2 Eloquent Elusor
    - [ ] etc (PLEASE, WRITE DOWN YOUR ROS VERSION HERE)

3. Which SBC(Single Board Computer) is working on TurtleBot3?

    - [x ] Raspberry Pi 3
    - [ ] Intel Joule 570x
    - [ ] etc (PLEASE, WRITE DOWN YOUR SBC HERE)

4. Which OS you installed in SBC?

    - [ x] Raspbian
    - [ ] Ubuntu MATE 16.04 or later
    - [ ] Ubuntu Server 18.04 or later
    - [ ] etc (PLEASE, WRITE DOWN YOUR OS)

5. Which OS you installed in Remote PC?

    - [ ] Ubuntu 16.04 LTS (Xenial Xerus)
    - [ ] Ubuntu 18.04 LTS (Bionic Beaver)
    - [ ] Linux Mint 18.x
    - [ ] Linux Mint 19.x
    - [x ] WINDOWS10 PC

6. Write down software version and firmware version

    - Software version: [x.x.x]
    - Firmware version: [x.x.x]

7. Write down the commands you used in order

    - successfully run SLAM and Navigation: all good.

8. Copy and Paste your error message on terminal

    - no error (except SLAM needed me to reverse the direction of Lidar on top of burger (opposite to build instructions): I presume this is a parmeter error on Windows PC distribution ?)...

9. Please, describe detailedly what difficulty you are in

    - Question: Does Turtlebot3 use EKF to improve Odometry ?

Your book "ROS Robot Programming" mentions EKF many times and says Odometry pose is improved by merging wheel sensors with IMU via EKF.

Having searched the code (e.g. Arduino) I see the "Calculate the odometry" section and find you seem to just take IMU for YAW, and wheel spin for distance.

I see AMCL for NAV,  and Cartographer works OK on Windows 10 PC... but no use of EKF (e.g. robot_localization ?).

Is this correct ? ... do you have EKF code / process that I have not found ?

Where is it please ?

If not used, why ?  .... do you recommend using robot_localization before feeding into SLAM ?

Many thanks

Michael Hallett

ROBOTIS-Will, 2020-07-21:
@MikeHallettUK 
Hi, TurtleBot3 does not use EKF for the odometry calculation. Please see the calcOdometry function.
https://github.com/ROBOTIS-git/opencr/blob/master/arduino/opencr_arduino/opencr/libraries/turtlebot3/examples/turtlebot3_burger/turtlebot3_core/turtlebot3_core.ino#L500

The reason being, there are situations when the odometry becomes incorrect due to various environmental condition. For Turtlebot3, 360 degree LIDAR can used to calibrate this error.

closed 2020-08-27 by MikeHallettUK

Rezenders, 2021-04-30:
@MikeHallettUK did you manage to do this?
> Understood.... but a more accurate odom from ekf combine of imu and wheel tick would still be a better starting point.

I am trying to use the `ekf_localization_node` node from [robot_localization](http://wiki.ros.org/robot_localization) package to merge the turtlebot ` /odom` data with an IMU but the output is rapidly converging to `/odom` (which is wrong). I inspected the `/odom` topic and I saw that the covariance is 0 for all fields, I wonder if this might be the problem. Did you need to set up different covariance values for odometry? If so, how did you do it?

MikeHallettUK, 2021-05-04:
No I did not use ekf.  With some testing I found the Turtlebot3 odometry to be accurate enough.  It uses wheels clicks for distance, and IMU for Yaw.

Rezenders, 2021-05-04:
Thank you for the answer. 
I am trying to map an environment and my turtlebot is traveling around 55m and I am getting around 4m of error, which is about 7.5%. Do you remember if you had this magnitude of error?

MikeHallettUK, 2021-05-05:
I get less odom error I think ~ 2 to 3% ... which I then correct with SLAM (e.g. cartographer) to get under 1%.

see issues with especially yaw @ [http://www.buzzdev.guru/p/turtlebot3-imu-and-odom-yaw-drifts-and.html](http://www.buzzdev.guru/p/turtlebot3-imu-and-odom-yaw-drifts-and.html).  The older IMU apparently was worse for drift.

Rezenders, 2021-05-05:
I checked now and I am not having this drift.
I have another question, since the odom yaw comes from the IMU, why the yaw from the /imu topic differs from the /odom?

MikeHallettUK, 2021-05-06:
/odom and /imu yaw should change the same... but at startup, or especially reset, can be initialised differently: see in the Arduino code.

Rezenders, 2021-05-06:
Understood. Thanks!