ROBOTIS-GIT/turtlebot3 #622: Cannot bringup Turtlebot3

garyzabel, 2020-07-19:
ISSUE TEMPLATE ver. 0.3.0

1. Which TurtleBot3 you have?
Burger

2. Which ROS is working with TurtleBot3?
ROS 1 Kinetic Kame

3. Which SBC(Single Board Computer) is working on TurtleBot3?
Raspberry Pi 3

4. Which OS you installed in SBC?
Raspbian
  

5. Which OS you installed in Remote PC?
Ubuntu 16.04 LTS (Xenial Xerus)
    

6. Write down software version and firmware version
Software version:  16.04

7. Write down the commands you used in order
$ export TURTLEBOT3_MODEL=$burger
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_remote.launch
$ catkin_ws/src/turtlebot3/turtlebot3_bringup$ roswtf

8. Copy and Paste your error message on terminal
$ turtlebot3_bringup turtlebot3_robot.launch

... logging to /home/pi/.ros/log/23b72b08-c950-11ea-a7bd-9cb6d0c599a3/roslaunch-raspberrypi-1019.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.122:33101/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /turtlebot3_core/baud: 115200
 * /turtlebot3_core/port: /dev/ttyACM0
 * /turtlebot3_core/tf_prefix: 
 * /turtlebot3_lds/frame_id: base_scan
 * /turtlebot3_lds/port: /dev/ttyUSB0

NODES
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.141:11311

process[turtlebot3_core-1]: started with pid [1029]
process[turtlebot3_lds-2]: started with pid [1030]
process[turtlebot3_diagnostics-3]: started with pid [1031]
[INFO] [1595116558.895200]: ROS Serial Python Node
[INFO] [1595116559.021631]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1595116561.250107]: Note: publish buffer size is 1024 bytes
[INFO] [1595116561.269499]: Setup publisher on sensor_state [turtlebot3_msgs/SensorState]
[INFO] [1595116561.294986]: Setup publisher on firmware_version [turtlebot3_msgs/VersionInfo]
[INFO] [1595116561.869394]: Setup publisher on imu [sensor_msgs/Imu]
[INFO] [1595116561.904754]: Setup publisher on cmd_vel_rc100 [geometry_msgs/Twist]
[INFO] [1595116562.043787]: Setup publisher on odom [nav_msgs/Odometry]
[INFO] [1595116562.067711]: Setup publisher on joint_states [sensor_msgs/JointState]
[... 97 lines cut ...]
 * /turtlebot3_lds->/turtlebot3_diagnostics (/scan)
 * /turtlebot3_core->/turtlebot3_diagnostics (/firmware_version)
 * /turtlebot3_core->/turtlebot3_diagnostics (/sensor_state)
 * /turtlebot3_core->/turtlebot3_diagnostics (/imu)

ERROR Errors connecting to the following services:
 * service [/turtlebot3_lds/get_loggers] appears to be malfunctioning: Unable to communicate with service [/turtlebot3_lds/get_loggers], address [rosrpc://192.168.0.122:47893]
 * service [/turtlebot3_lds/set_logger_level] appears to be malfunctioning: Unable to communicate with service [/turtlebot3_lds/set_logger_level], address [rosrpc://192.168.0.122:47893]
 * service [/turtlebot3_diagnostics/set_logger_level] appears to be malfunctioning: Unable to communicate with service [/turtlebot3_diagnostics/set_logger_level], address [rosrpc://192.168.0.122:41929]
 * service [/turtlebot3_diagnostics/get_loggers] appears to be malfunctioning: Unable to communicate with service [/turtlebot3_diagnostics/get_loggers], address [rosrpc://192.168.0.122:41929]
 * service [/turtlebot3_core/set_logger_level] appears to be malfunctioning: Unable to communicate with service [/turtlebot3_core/set_logger_level], address [rosrpc://192.168.0.122:37569]
* service [/turtlebot3_core/get_loggers] appears to be malfunctioning: Unable to communicate with service [/turtlebot3_core/get_loggers], address [rosrpc://192.168.0.122:37569]

9. Please, describe detailedly what difficulty you are in
I've been struggling with this problem for several days. I'm trying to bringup Turtlebot 3 Burger for the first time. At first I thought that there might be a hardware problem with the OpenCr board. But I was able to update the firmware, so I assume that's not the difficulty. I tried removing the Turtlebot 3 packages and installing them again, but that did not help.I can't see anything obviously wrong with the description.launch.xml file:

<launch>
  <arg name="model"/>
  <arg name="urdf_file" default="$(find xacro)/xacro --inorder '$(find turtlebo$
  <param name="robot_description" command="$(arg urdf_file)" />
</launch>

Maybe the problem has something to do with xacro/urdf?

I'm at my wit's end with this one. I would deeply appreciately any help anyone is able to give.

garyzabel, 2020-07-19:
For some reason I can't get the launch file code to appear in its entirety. I'll give it one more try here.

 GNU nano 2.5.3          File: description.launch.xml                          

<launch>
  <arg name="model"/>
  <arg name="urdf_file" default="$(find xacro)/xacro --inorder '$(find turtlebo$
  <param name="robot_description" command="$(arg urdf_file)" />
</launch>

Chinmay922, 2020-07-23:
Hey, I tried using the same commands as you have shown in this issue and see that if I do the following checks it works fine for me but to get to your problem could you do the following,
1. Can you share the complete set of commands you used on your terminal while performing bring up? 
2. Did you run roscore on your remote PC terminal along with the bring up launch file on the turtlebot3 terminal?
3. Can you also check if the IP address in your bash.rc file for your turtlebot and remote PC is in the correct format as shown in the picture below?
I see a few things to correct in your case and will probably get to the solution if you send me the commands and check for the above points.  

Steps to check - 
1. Run roscore command on your remote pc terminal 
2.  ssh into raspberry pi using ssh command - https://www.raspberrypi.org/documentation/remote-access/ssh/
 Note - once you are in the raspberry pi terminal use the following commands
a)  I can see an invalid tag on your turtlebot3 description 
$ export TURTLEBOT3_MODEL=$burger     **Incorrect**
 $ export TURTLEBOT3_MODEL=burger  **Correct**

b)
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_remote.launch   **incorrect**
pi@raspberrypi:~ $ roslaunch turtlebot3_bringup turtlebot3_robot.launch   **correct**

garyzabel, 2020-07-25:
Thanks so much for your help! I caught the export TURTLEBOT3_MODEL=$burger error and corrected it. I also was having problems with the  serial port assignment, but was able to correct that. After some other tweaks that I failed to keep track of, I was able to use bringup successfully. But now I'm back in Robot Hell trying to get SLAM up and running: a Timeout problem. If I can't solve it soon, I'll post another entry on Issues. I really appreciate your help with this. I retired 2 years ago from teaching philosophy and decided recently to try to learn Ros. I need all the help I can get.

closed 2020-07-25 by garyzabel

Chinmay922, 2020-07-26:
Good to hear that! Let me know if you need any help further. :)