ROBOTIS-GIT/turtlebot3 #624: Release for ROS noetic

Rayman, 2020-07-24:
Hey everyone,

Are there plans for releasing turtlebot3 to ROS noetic? After a quick search I could not find nothing about this.

Thanks!

ROBOTIS-Will, 2020-07-27:
Hi,
Yes, we are preparing for Noetic release. I've been working on some dependencies recently for TurtleBot3 package update.
You should be able to see Noetic in August.
Thank you.

ROBOTIS-Will, 2020-08-25:
@andrestoga Thank you for pointing it out. We will start working on TB3 packages this week including Noetic and Foxy.
There are some other hanging issues such as raspicam node, but we'll release first and return for them later.

estshorter, 2020-09-17:
@ROBOTIS-Will Do you have any news on foxy and noetic support?

ROBOTIS-Will, 2020-09-23:
Hi, @estshorter 
Thank you for your interest in this.
We're currently working/testing on Noetic with Raspberry Pi3B+, but having some issues with Ubuntu crashing(#593).
I'll try to release the Noetic as soon as possible.
In case you want to try it out first with `noetic-devel` branch, please refer to #639 for the Noetic OpenCR firmware as there is an updated message.

My test TB3 environment for Noetic is :
- [Ubuntu Mate 20.04(64bit) for RPi3B+](https://ubuntu-mate.org/download/arm64/focal/)
- ROS Noetic

Please note that I had to set the swap area of 2GB in order to properly build the source.

pieterwolfert, 2020-09-23:
Hi @ROBOTIS-Will, I'm currently also testing out Ubuntu 20.04 server installed on a Raspberry Pi3B+ in combination with ROS Noetic. After updating the OpenCR software it gave a message: failed to set dtr/rts, does that hurt? I also installed the packages not specifically for ROS Noetic (by cloning turtlebot3 and related packages in catkin_ws/src), they seemed to build, but I take it's better to use the packages from the noetic-devel branch?

pieterwolfert, 2020-09-23:
It seems to be working under Ubuntu 20.04 on a RPi3B+ in combination with Ubuntu 20.04 on the master pc, both with ROS Noetic, as messages are passed on. Yet, if I try the remote teleop, the commands are published (and arrive on the Pi), but the wheels don't turn..

ROBOTIS-Will, 2020-09-25:
Hi @pieterwolfert,
The bug fixed 20.04 on a daily build is available from the link below and we are currently testing if it works fine.
http://cdimage.ubuntu.com/ubuntu-server/focal/daily-preinstalled/current/

Since there are some changes in Noetic, I'd recommend to use Noetic packages instead of cloning precompiled pacakges from different ROS version.

I haven't encountered the **failed to set dtr/rts** yet, but if any failure occurred during the OpenCR firmware uploading, simply retry the upload process or try under the [Recovery Mode](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#recovery-mode).

ROBOTIS-Will, 2020-12-01:
This issue thread will be closed as TurtleBot3 Noetic has been released and there is no recent activity. Please feel free to reopen if needed.

closed 2020-12-01 by ROBOTIS-Will